prusa3d/PrusaSlicer #6610: Flash Firmware fails

JoTiZ, 2021-06-11:
### Version
2.3.1 (Snap from Ubuntu Store - *error*)
2.3.0-rc1+linux-x64 (Appimage from github - *working*)

### Operating system type + version
Ubuntu 20.04.2 LTS x86_64
Had the same problem on Manjaro 21.0.5 a few minutes ago. Here it was PrusaSlicer 2.3.1, from Snap, too.

### 3D printer brand / version + firmware version (if known)
Artillery Sidewinder X1 + Marlin 2.0.8.2

### Behavior
First of all: I am a groupmember of dialout under Ubuntu. The following results are all from my Ubuntu machine.

Started Prusa Slicer 2.3.1 and clicked on "Flash printer firmware"
Selected the new .hex file and selected the port 
*clicked on flash*

---------------------------------------------------- Result PrusaSlicer 2.3.1 -----------------------------------------------
avrdude-slic3r -v -p atmega2560 -c wiring -P /dev/ttyUSB0 -b 115200 -D -U flash:w:0:/home/doe/Desktop/firmware.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on May  2 2021 at 19:50:12
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : /dev/ttyUSB0
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
avrdude-slic3r: ser_open(): can't open device "/dev/ttyUSB0": Operation not permitted
avrdude-slic3r: Could not open port: /dev/ttyUSB0

avrdude-slic3r done.  Thank you.
---------------------------------------------------End error result-------------------------------------------

Same machine, same login, same .hex-file only other software version:

---------------------------------------------------- Result PrusaSlicer 2.3.0-rc1 -----------------------------------------------
avrdude-slic3r -v -p atmega2560 -c wiring -P /dev/ttyUSB0 -b 115200 -D -U flash:w:0:/home/doe/Desktop/firmware.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Dec 17 2020 at 20:35:16
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : /dev/ttyUSB0
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: reading input file "/home/doe/Desktop/firmware.hex"
avrdude-slic3r: writing flash (135686 bytes):
avrdude-slic3r: stk500v2_command(): command failed

Writing | ################################################## | 100% 24.94s

avrdude-slic3r: 135686 bytes of flash written
avrdude-slic3r: verifying flash memory against /home/doe/Desktop/firmware.hex:
avrdude-slic3r: load data flash data from input file /home/doe/Desktop/firmware.hex:
avrdude-slic3r: input file /home/doe/Desktop/firmware.hex contains 135686 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 16.48s

avrdude-slic3r: verifying ...
avrdude-slic3r: 135686 bytes of flash verified

avrdude-slic3r done.  Thank you.
---------------------------------------------------End of good result-------------------------------------------

rtyr, 2021-06-11:
Flash printer firmware option is meant for Prusa printers only.

JoTiZ, 2021-06-12:
Thank you for this information - closed

closed 2021-06-12 by JoTiZ