prusa3d/PrusaSlicer #6611: When using MMU painting feature, parts of the model are missing in sliced version.

ssill2, 2021-06-11:
### Version
_Version of PrusaSlicer used goes here_
2.4.0-alpha from master branch as of june 11th 2021, revision https://github.com/prusa3d/PrusaSlicer/commit/e17b57d6bdaf7471ceeb2509d9f774b9f7648398

_Use `About->About PrusaSlicer` for release versions_
https://github.com/prusa3d/PrusaSlicer/commit/e17b57d6bdaf7471ceeb2509d9f774b9f7648398
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
e17b57d6bdaf7471ceeb2509d9f774b9f7648398

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
MK3S+ + MMU2S

### Behavior
* _Describe the problem_
When using the MMU painting feature, parts of the model disappear when sliced.  See attached.  Part of the lettering is missing on slicing.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
Open the attached .3mf and slice.

* _Expected Results_
it's expected that the lettering would remain intact in the painted color.

* _Actual Results_
Part of the lettering is simply missing in the sliced version.  No need to print, you can see it after the slicing has happened.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[LCD-cover-ORIGINAL-MK3.zip](https://github.com/prusa3d/PrusaSlicer/files/6640041/LCD-cover-ORIGINAL-MK3.zip)

hejllukas, 2021-06-15:
Thank you for reporting. I looked at it, and the issue is that in places where the text is embossed, the model is quite thin, which is okay for single nozzle printing.
But in your case, this thin place is split in half into two thinner parts (shown in the attached picture) because each side is painted with a different color. These newly formed parts become too thin to be printed, which is why they disappear. 
[image]

However, if you turn on "Detect thin walls" (Layers and perimeters->Quality), even these very thin parts will be printed, and they will not disappear in preview.

ssill2, 2021-06-15:
ah thank you very much!  I'll close this issue then.  Much appreciated.

ssill2, 2021-06-15:
So interestingly.  I took your suggestion and turned on the detect thin walls and attempted to reslice.  and it crashes the prusaslicer lol  It just exits with no warning.

ssill2, 2021-06-15:
Here's the .3mf with parts reoriented but simply slicing makes the program crash.

[LCD-cover-ORIGINAL-MK3 - crashes.zip](https://github.com/prusa3d/PrusaSlicer/files/6655548/LCD-cover-ORIGINAL-MK3.-.crashes.zip)

ssill2, 2021-06-15:
latest commit to master seemed to fix the crashing

ssill2, 2021-06-15:
I'll close this again since it appears fixed.  thank you!

closed 2021-06-15 by ssill2