prusa3d/PrusaSlicer #6614: BUG: auto arrange with disable items - move all items outside table

KubaMiszcz, 2021-06-12:
### Version
PrusaSlicer 2.3.1.+win64
### Operating system type + version
Win10 64bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
when I have multiple items and set few of them hidden, auto arrange move all items outside the table, 

then when i manually move visible items to the center of table it automatically recenter ALL items along with their center of mass(?) to the center of table

desired result:
put hidden items outside table, and arrange and center visible item to the center of the table (like in version 2.3.0)

[image]
[image]

rtyr, 2021-06-14:
Please provide your 3MF project file (before arrange).

tamasmeszaros, 2021-07-16:
Hi @KubaMiszcz ! I tried to reproduce this issue on multiple platforms but I couldn't. Please provide some 3mf project files, or we will probably close this issue as it's not reproducible on our side.