prusa3d/PrusaSlicer #6615: nlop_cxx library not found - CentOS 7

yoyo42, 2021-06-13:
I'm building 2.4.0_alpha on CentOS 7 (commit e17b57d6bdaf7471ceeb2509d9f774b9f7648398) and had an issue with finding the NLopt library:

CMake Error at cmake/modules/FindNLopt.cmake:124 (message):
  Unable to find requested NLopt installation: Cannot find NLopt library
  'nlopt'.  NLopt not found in system directories (and environment variable
  NLOPT is not set).

In cmake/modules/FindNLopt.cmake the library name if NLopt_DIR is specified is "nlopt_cxx" (correct on my system) but if NLopt_DIR is not specified then it's only "nlopt" which is not found.

I fixed this by changing cmake/modules/FindNLopt.cmake:31 to be "set(_NLopt_LIB_NAMES "nlopt_cxx")" but I don't know enough about cmake or why it might be different in the two cases to do a pull request.

A previous closed issue (https://github.com/prusa3d/PrusaSlicer/issues/3186) implies that it used to be called "libnlopt" on the file system on their Windows NLopt install, so maybe it was originally made to match that?

(That issue also suggests that cmake should be pulling dependencies itself, but I've seen no sign of that, I've had to collect lots of them manually.)

tamasmeszaros, 2021-07-19:
Hi @yoyo42  
I would not recommend building PS with dependencies installed from distro packages in case of Centos7. NLopt is an exception as its even newer than what we use, but the other crucial libraries are either missing or hopelessly outdated.

That being said, Centos7 is a great distro to build PS with the dependency download scripts and linking statically as described in our [tutorial](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md) 

Centos7 also ships an older gcc which is not recent enough to build PS. You can install newer compilers however, with the `devtoolset-*` packages and then use a fairly cutting edge compiler suit.  `devtoolset-7` includes a recent enough compiler, but you can go with `devtoolset-10` as well.

yoyo42, 2021-08-15:
Interesting, thanks for the info, I'll try devtoolset-10. The tutorial you link could do with rephrasing to match what you just said  - it says this right at the start: 

"generally versions available on conservative Linux distros such as Debian stable or CentOS should suffice."

tamasmeszaros, 2021-08-18:
Thanks for the remainder @yoyo42, I'll take a look at the tutorial. Things just change too quickly to keep it up to date. It is very likely that you could build with distro packages around v2.1 - v2.2. But we don't check too often as almost everything is built statically and only a few libs are linked to PS from system. I've linked the tutorial mainly to aid you in the dependencies build procedure which should be reasonably well described there.

Please make a comment if you have succeeded with the build, so we can then close this issue.