prusa3d/PrusaSlicer #6620: Slicing generates a large area of bridge-infill in an empty area.

stepheneb, 2021-06-14:
### Version
2.3.1

### Operating system type + version
macOS: 11.4

### 3D printer brand / version + firmware version (if known)

Problem in slicer.

### Behavior
Slicing generates a large area of bridge-infill in an empty area.

Modeled a threaded cap using FreeCad. Cap can have 3, 2.5, 2.0, and 1.5 threads. The slicer issue only appears with the cap with 1.5 threads (the shortest).

The bridge-infill slicing artifact does not appear:

1. When taller models with a larger number of threads are sliced.
2. When the original stl is processed through netfabb and the netfabb model is sliced.

Seems similar to issue: #3413

Object on left is my original model. Object on right is the same model processed by netfabb. Have also attached 3mf file with both models.

[image]

PrusaSlicer does not appear to display errors for the original stl file.

[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs

Attached is 3mf file saved from PrusaSlicer: `threadedcap60-1.5.3mf.zip`

Two objects:

1. threadedcap60-1.5.stl
2. threadedcap60-1.5_fixed.stl

Second object: `threadedcap60-1.5_fixed.stl` is original file processed through netfabb.

[threadedcap60-1.5.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6651754/threadedcap60-1.5.3mf.zip)

stepheneb, 2021-06-15:
These reports generated by the running the **Check All** command in the Blender **3D Print Toolkit** addon might have clues.

Two biggest differences I notice are the original file has 434 **Non Manifold Edges** and 61 **Zero Edges**. I'll investigate what that means. These counts are both 0 in the file run through netfabb.

Report for the original stl file.

[image]

Same report for the stl file run through netfabb:

[image]

ETI4K, 2026-09-21:
Having a very similar problem.
Part is modelled in Inventor 2020 on Windows 11, then saved as a .stp file.  When sliced in 2.9.6, open areas of the interior have solid bridging(?)/infill.
The solid infill appears at and between discontinuities of the solid that exist only in the slicer.

I'm not sure how to provide you with the 3mf, as it's not a permitted file type.
The Nozzle is sectioned in Inventor in the image titled: "Nozzle - Sectioned".
The other images are as-sliced.
In "Screenshot 2026-09-21 060613.png", I cut the part in the slicer just to see what the slicer was doing.

[image]
[image]
[image]
[image]