prusa3d/PrusaSlicer #6622: Specify draft shield height

huskyte, 2021-06-16:
### Version
PrusaSlicer
Version:   2.3.0+
Build:     PrusaSlicer-2.3.0+-202101111304

### Behavior
* Draft Shield activated
PrusaSlicer gives a nice draft shield that is exactly as high as the model itself. At least in my experience the printing of the draft shield can fail over time with ASA/ABS as deformation takes place resulting in little fibres or whatever that sooner or later will block the hot end fan. 

Proposed mitigation: The purpose of the draft shield is to hold a microclimate for the model while printing which isn't necessary over the whole print height. Especially for the last layers (could be something like the last 30% of the height) it isn't needed at all. So, here it is just a risk without benefit.

I propose a simple setting in PrusaSlicer specifying the height either in % of the total model height or as absolute mm ("Height draft shield, mm or %")

_Is this a new feature request?_ Yes

rtyr, 2021-06-16:
It's already possible. Set skirt height to prefered value (number of layers) and don't enable "draft shield" option (it only extends skirt height to the max).

huskyte, 2021-06-16:
Oh, hell, thx. I just realized that this was really the case. thx for the hint. Looking at it a "%height" or/and "mm max"(? a bit misleading, skip over a certain height or something similar) would be good, but going for layers is ok to handle.

wavexx, 2021-08-14:
I think it's no longer possible in the current master, where the skirt is always placed outside of the brim.

rtyr, 2021-08-14:
That's what limited draft shield is for.

wavexx, 2021-08-14:
Yeah, noticed after playing with it, sorry!

I like the dedicated option, but I almost feel like that maybe the text in option group should now be "Skirt/shield options", since they apply to both to make it obvious?