prusa3d/PrusaSlicer #6623: I have a question about the speed of the first layer after the raft.

heavy2119, 2021-06-16:
Thanks for fixing the 2.4.0 version raft to work properly.

This modified version 2.4.0 has been completely modified to be comparable to Cura and Simply 3D.

However, speed in the first layer after the raft is most urgent.

I know that the first layer speed correction after raft is also reflected.

I arbitrarily modified the speed of the first layer and printed it out, and it was perfect.

When will the first layer speed change version be fixed?

bubnikv, 2021-08-31:
Fixed with b29c0ead7dfc1a997cad012b9f04e84cec1beb9a
   Implemented configurable speed and acceleration settings for the first
    object layer over raft interface:
    "first_layer_speed_over_raft", "first_layer_acceleration_over_raft".
    Fixes I have a question about the speed of the first layer after the raft. #6623
    Fixes Layer After Raft Is Not Considered First Layer! #6166

The fix will be released with PrusaSlicer 2.4.0-alpha1.

closed 2021-08-31 by bubnikv

StevenInc, 2023-06-29:
What is Speed of object first layer over raft interface? Where is the documentation for this setting please.