prusa3d/PrusaSlicer #6625: PrusaSlicer inserts unwanted g-code

Bimbovadasz, 2021-06-16:
### Version
Version 2.3.1+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
SecKit SK-go + Marlin 2.0.7.2, but it's not really relevant. 

### Behavior
I am trying to use custom g-codes in PrusaSlicer for a tool changer build. The sliced g-code looks a bit messy because of the indentations and conditions, but it works no problem. The issue is that the slicer inserts M109 and M106 after and even into my custom g-codes, even though it is already there a few lines back. 

Part of my TC g-code:
[image]
The sliced result with an unwanted M109:
[image]
I thought the conditions and indentations confuses the slicer, so I put a M109 directly before the slicer inserts it to test if it helps, but the slicer still puts in its own line for some reason.
My test g-code:
[image]
The result:
[image]

Is there a way to get rid of this or get around it? 

Thanks for the help!

neophyl, 2021-06-16:
What I’m seeing from your pics is you are inserting M109 T1 S210 and the slicer is automatically adding in M109 S210 T1. Note the difference. Personally I’d be testing if the order of the S and T values makes a difference.

Bimbovadasz, 2021-06-16:
Unfortunately it does not make a difference
[image]

closed 2026-09-29 by prusaslicer-bot[bot]