prusa3d/PrusaSlicer #6627: Feature Request: Arrange all on bed only

napter, 2021-06-16:
### Version
2.3.1

### Behavior
As part of my workflow I will usually load many models into the workspace. I will then drag items on to the bed that I want to currently print. I would like the option to arrange models but just the ones on the bed. Currently this can be done in a very tedious fashion by ctrl + clicking each one but when I have 20+ small models on the bed this is difficult (and one miss click and everything gets unselected). Instead an option to arrange all on bed would be great. 

In fact, perhaps this should even be the default behavior as typically everything you want to print is on the bed. If you drag it off the bed you probably don't want to currently print it. It will give better control over bed setup - for example, currently if you put too many things on the bed, the arrange will move some off. If you didn't like which items it chose to print first you would be able to drag an item it put on the bed, off and put another back on and click arrange again to create a bed with your chosen items.

rtyr, 2021-06-16:
Select models as mentioned in https://github.com/prusa3d/PrusaSlicer/issues/6626, then use SHIFT+A (arrange selection).

Closing.

closed 2021-06-16 by rtyr

edwardhorsford, 2022-01-11:
I came here to raise this same issue - but thankfully found this closed one.

I'm glad this is (somewhat) supported, but would like to vote for the requested feature getting built as well.

Reasons:
* Selecting the items first was non-obvious
* Selecting items individually could be rather time consuming if you have lots of parts - no drag selections currently.
* For me personally, I want arrange to always only include items on my bed. So I'd much prefer it as a global default / setting than the current behaviour.