prusa3d/PrusaSlicer #6630: macOS Build

ahousseini, 2021-06-18:
### Version
All releases

### Operating system type + version
All macOS releases

### Behavior
* We use [AutoPkg](https://github.com/autopkg) to build packages for distribution and I noticed the keys in the [Info.plist](https://github.com/prusa3d/PrusaSlicer/blob/master/src/platform/osx/Info.plist.in) are not standard.

These 3 keys need to be changed:

1. `CFBundleVersion`: Should be for the latest release `202104161356` instead of `PrusaSlicer-2.3.1+x64-202104161356`
2. `CFBundleShortVersionString`: Should be for the latest release `2.3.1` instead of `PrusaSlicer PrusaSlicer-2.3.1+x64-202104161356`
3. `CFBundleIdentifier`: Should be `com.prusa3d.slic3r` instead of `com.prusa3d.slic3r/`

Here are some good resources especially about `CFBundleVersion` and `CFBundleShortVersionString`:
- https://developer.apple.com/library/archive/technotes/tn2420/_index.html
- https://stackoverflow.com/questions/19726988/what-values-should-i-use-for-cfbundleversion-and-cfbundleshortversionstring

It would be very helpful to adjust these keys! Thanks!

hildahlp, 2022-04-28:
This is an issue for us as well. The two keys (CFBundleVersion and CFBundleShortVersionString) identified by ahousseini should conform to macOS developer guidelines to be useful for version management. CFBundleIdentifier seems to have been corrected and is correct in 2.4.2

iammattcoleman, 2026-09-03:
This is still an issue in the 3.0.0 alpha. From the 3.0.0-alpha11 build:
```
  <key>CFBundleShortVersionString</key>
  <string>PrusaSlicer PrusaSlicer-3.0.0-alpha11</string>
  <key>CFBundleIdentifier</key>
  <string>com.prusa3d.slic3r/</string>
...
  <key>CFBundleVersion</key>
  <string>PrusaSlicer-3.0.0-alpha11</string>
```

1. `CFBundleShortVersionString` should be set to just "3.0.0", not "PrusaSlicer PrusaSlicer-3.0.0-alpha11". The [CFBundleShortVersionString documentation](https://developer.apple.com/documentation/bundleresources/information-property-list/cfbundleshortversionstring) states:
   > The required format is three period-separated integers, such as 10.14.1. The string can only contain numeric characters (0-9) and periods.

2. `CFBundleIdentifier` cannot contain the trailing slash that it currently has. From the [CFBundleIdentifier documentation](https://developer.apple.com/documentation/bundleresources/information-property-list/cfbundleidentifier):
   > The bundle ID string must contain only alphanumeric characters (A–Z, a–z, and 0–9), hyphens (-), and periods (.).

3. `CFBundleVersion` should be set to just "3.0.0", not "PrusaSlicer-3.0.0-alpha11". The [CFBundleVersion documentation](https://developer.apple.com/documentation/bundleresources/information-property-list/cfbundleversion) says:
   > This key is a machine-readable string composed of one to three period-separated integers, such as 10.14.1. The string can only contain numeric characters (0-9) and periods.

This seems like it should be an easy fix, but it would require a 3-segment version number being available in `version.inc`. [These comments](https://github.com/prusa3d/PrusaSlicer/blob/version_3.0.0-alpha11/version.inc#L1-L2) imply that there's a build script that updates the file, but I wasn't able to find any scripts that touch `version.inc` in the repo. Is that script external to the repo (if so, where does it live?) or is that possibly an outdated comment?