prusa3d/PrusaSlicer #6635: Sidewinder X1 stops dead after priming with default printer-profile.

LeonFretter, 2021-06-19:
### Version
2.3.1

### Operating system type + version
Debian 10

### 3D printer brand / version + firmware version (if known)
Artillery Sidewinder X1 (stock)
Marlin-2.0.5.3

### Behavior
The printer primes, then moves to the middle as if to start the print, moves a tiny bit then stops dead.
Pressing "Pause", then "Resume" makes it continue (It needs a short time before the "Pause" is executed though).

I isolated this problem to the custom gcode of the default "Sidewinder X1 Printer-Settings". How? I changed around
everything I could, and in my most recent try only inserted the default gcode from the non-prusa slicer into the custom-gcode section.
Namely:

Start G-code:
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

End G-code:
M104 S0 ; turn off temperature
G28 X0  ; home X axis
M84     ; disable motors

and before-layer, after-layer and tool-change empty.

With that the printer printed fine again so the problem must lie (or at least be partially attributed) to the custom g-code in the 
default sidewinder printer-profile.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Unfortunately I don't have such a button. 
I only have the option to export as "AMF", shall I provide that instead?

rtyr, 2021-06-19:
Probably same issue as https://github.com/prusa3d/PrusaSlicer/issues/6580 (caused by M900/M200 commands not supported by stock fw).

M900/M200 were removed in the latest configuration update (0.0.3). It is currently used only with BLTOUCH variants.

Make sure you are using latest configuration update.

[image]

rtyr, 2021-06-19:
If you have older configuration, you should see update notification in the bottom right corner.

LeonFretter, 2021-06-19:
I have the current version (0.0.3) and I am using the standard profile (not BLTOUCH), so this seems to me to be an unrelated problem.

rtyr, 2021-06-19:
Please provide your 3mf project file (file -> save project as..) or at least the g-code.

rtyr, 2021-06-19:
@SzabolcsHornyak FYI

LeonFretter, 2021-06-19:
Ah okay, thought that 3mf stuff was supposed to be under export.
My printer currently prints something. I will let it finish, then create a project with the aforementioned settings, reassure that it creates the problem, then upload the 3mf-file here.

LeonFretter, 2021-06-19:
Okay, I couldn't get a buggy project running right now, but found a gcode-file which still has the buggy code on it, and sure enough there is the M900 and M200 on it.
Sorry for that, could have spotted that earlier (Must have had the old pre 0.0.3 Config then, and updated in the midst of the bug-tracing without noticing it) - when you asked for it earlier I just checked the config :/
Yeah, sorry, was a long, hot day... Closing this as a duplicate of #6580 then

closed 2021-06-19 by LeonFretter