prusa3d/PrusaSlicer #6636: Layer shifting Version 2.3.1+win64

6XoCtujg2C0gne, 2021-06-19:
### Version
Version 2.3.1+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10
PrusaSlicer
Version:   2.3.1+win64
Build:     PrusaSlicer-2.3.1+win64-202104161404

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 12,741MB
OpenGL installation
GL version:   4.6.0 NVIDIA 457.30
Vendor:       NVIDIA Corporation
Renderer:     GeForce GT 710/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
[... 297 lines cut ...]
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Highly modified CR10S
#define CONFIGURATION_H_VERSION 020008

### Behavior
I've noticed several times when slicing if I use .20 resolution I get a good print. However, if I change the resolution
to anything less than .20 (eg. .18 or .16) I get shifted layers.
[image]
It appears that the shifting began around Layer 77 or 78.
It's worth noting that this has happened at least 3-4 times with this version.

#### Project File (.3MF) where problem occurs
[Main Plate Mount.zip](https://github.com/prusa3d/PrusaSlicer/files/6681043/Main.Plate.Mount.zip)
[image]

rtyr, 2021-06-20:
The g-code generated from your 3mf project is OK, so is the test print (I only added G92 command to simulate your physical z-offset). I don't see any issue related to PrusaSlicer.

[image]

6XoCtujg2C0gne, 2021-06-20:
I did notice some inaccuracies with the model in the slicer but I didn't catch initially,
I fixed the issues which you can see in PrusaSlicer after slicing.  I re-sliced but, I used S3D
instead.
I have seen this happen more than once as I indicated however, I'll close this one and if it happens again I'll chime in.
Thanks for looking into this!

closed 2021-06-20 by 6XoCtujg2C0gne