prusa3d/PrusaSlicer #6637: Feature request : List results of combining nozzle size + layer height, etc in a table

imnotme, 2021-06-19:
### Version
2.3.1

### Describe the feature request
It should be great to generate a table with all the results from different user-selected combination of nozzle size, layer height, perimeters, infill, etc when slicing a model

### Expected Results

Import your .STL

Open the new and awesome "Calculate print" menu

You choose the combinations with a step by step wizard, or by ticking the elements from a list.

You click on generate, and voilà, a table like this is generated (real data from my last print) :

| Nozzle | Layer Height | Print time | Material consumption |
| --- | --- | --- | --- |
| .25 | .10 | 6h53 | 6.56m |
| .25 | .15 | 4h46 | 6.56m |
| .4 | .15 | 2h52 | 6.71m |
| .4 | .20 | 2h12 | 6.57m |
| .6 | .20 | 1h53 | 7.79m |
| .6 | .35 | 1h09 | 8.54m |

### Why ???
Because with the previous table I can now choose the settings I want, for example

- Woodfill, PCCF, Flex are expensive,  I will choose the best Layer height/Material consumption couple for this print (.4/.20)
- A functionnal part I need quickly, I will choose the best print time/layer height combination, if PCCF material cost will also count (.6/.20 or  .4/.20 that cost 19minutes more to print but 1.2m less of material :dollar: )
- I need details but don't want to print for 3 days, I will choose the best layer height/print time for my .25 nozzle (almost 2h gained  just by increasing layer height by 0.05, we have a deal here :smiley:  )

I could also affect print time by adjusting numbers of perimeters, etc, like theses
- Infill %
- Number of perimeters
- Infill type
- ...

Next step could be some kind of AI wizard that analyze material cost, recognize details if present, ask one or two questions about if we want details or speed, then propose the best settings for the imported STL.

### Is this a new feature request?

Yes, but at some points #4397 can be related

murk-sy, 2021-06-20:
This is a pretty niche request that is probably beyond the scope of the slicer itself. But you could probably automate it relatively easily with commandline.

https://github.com/prusa3d/PrusaSlicer/wiki/Command-Line-Interface
https://manual.slic3r.org/advanced/command-line

1) Load STL and settings (not sure if you can just prepare a 3mf and load all defaults from that, might need to do .stl and load the ini file by path)
2) Generate gcode for every nozzle size and layer height using ```--layer-height``` and ```--nozzle-diameter``` and save them to a file with a name that indicates settings
3) Iterate through every .gcode file to find ```; total filament used [g] =``` and ```; estimated printing time (normal mode) =```
4) Output data in a readable format

imnotme, 2021-06-22:
> This is a pretty niche request that is probably beyond the scope of the slicer itself. But you could probably automate it relatively easily with commandline.

We are talking about print optimization, I'm not sure this is beyond of the slicer itself, as the print cost & others are already embedded.

And yes, _relatively_ is the word to use here :smile: I already tried a quick n dirty solution like this, but It was a little more complicated than expected, so feature request and if it's never implemented, I will spend time to script it.

Thanks for your attention anyway

closed 2026-09-29 by prusaslicer-bot[bot]