prusa3d/PrusaSlicer #6638: Layer Hight remains at 0.3 whatever I set it to.

3Dietrich, 2021-06-20:
### Version
2.3.0+

### Operating system type + version
PrusaSlicer
Version:   2.3.0+
Build:     PrusaSlicer-2.3.0+-202101111304

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 10.16 (Build 20F71)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   2.1 Metal - 71.6.4
Vendor:       Apple
Renderer:     Apple M1
GLSL version: 1.20

### 3D printer brand / version + firmware version (if known)
i3Mk3s
### Behavior
#### _Describe the problem_
I'm not sure if I don't realize the obviuos (wouldn't be the first time) :
**I can not change layerhights.** Well: Ich can change it but it doesn't show it.
#### _Steps needed to reproduce the problem_
I took a PrusaSlicer project from 2019 ([Original from 2019](https://github.com/prusa3d/PrusaSlicer/files/6682114/hand.fan.all.3mf.zip))
and made some changes. I noticed the following problem: no matter which layer height I set, it always stays at 0.3mm.
([actual edited with the problem](https://github.com/prusa3d/PrusaSlicer/files/6682122/hand.fan.Rippchen_haa.3mf.zip))

#### _Expected Results_
Slicer should show the set layer heights (e.g. 0.1mm --> should show layers 0.2, 0.3, 0.4..)
#### _Actual Results_
Slicer shows always 0.3mm layer hight (e.g. 0.1mm --> it shows 0.2, 0.5, 0.8..)

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

https://user-images.githubusercontent.com/4608796/122671576-ac0a8600-d1c7-11eb-89ff-08378b5215c9.mov

#### Project File (.3MF) where problem occurs:
[actual edited with the problem](https://github.com/prusa3d/PrusaSlicer/files/6682122/hand.fan.Rippchen_haa.3mf.zip)
[Original from 2019](https://github.com/prusa3d/PrusaSlicer/files/6682114/hand.fan.all.3mf.zip)

3Dietrich, 2021-06-20:
ah! I found the problem (and this turns the issue in a request): there was something set in "Variable Layer Height". This setting kept all the time active without warning. A Warning would be helpful (e.g. close to the 'standard' Layerhight an indicator or warning 
"Attention! LH additionally changed by _Variable LH_" 
or something like that.

rtyr, 2021-06-20:
We added a notification to the object list, it is already on master. Closing.

[image]

closed 2021-06-20 by rtyr

3Dietrich, 2021-06-20:
fuck. I sit hours to get it clear and understandable and wrote the bugreport.
And you throw it directly in the trash :(

rtyr, 2021-06-20:
Did you see the screenshot I posted? Your request is already implemented...

3Dietrich, 2021-06-20:
yes I saw it.
Your demand makes me bear the result better: You have 'my' suggestion already built in. Excellent!

rtyr, 2021-06-20:
ok :-)

bubnikv, 2021-06-26:
We implemented this kind of notification for various other features that
may be applied to the model, but are not visible at the first sight. For
example, the paint-on supports or paint-on seam. This will be a rule moving
forward.

You see we are thinking hard making the application more usable :-)