prusa3d/PrusaSlicer #6639: Freezes and quits when changing extruder on multi-body object

evodevosys, 2021-06-21:
### Version
2.3.1+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa MKS3+ with MMU2S

### Behavior
* I am printing in multi color mode. I import multiple STL files (exported as one body per file) and choose YES when Prusa Slicer asks me if I want to consider them parts of a single object.

The extuder is assigned to the parts in the file name order (cycling through the extruders).  I want to change which part has which extruder so that I have the right color for each part.
I start going through the parts and changing the extruder, but usually at some point the software will freeze and quit.
[SlicerProblem.zip](https://github.com/prusa3d/PrusaSlicer/files/6688972/SlicerProblem.zip)

* _Steps needed to reproduce the problem_
*Double click on the extruder to for a part to change it.  In the video, it quit right away on the first one.  Usually it quits later on so by saving intermittently and restarting from where I left off, I can get through it all after several rounds.

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-06-25:
I was able to reproduce it, thank you. 

It seems it happens only if you use keyboard to navigate in the extruder selection (arrows or direct number input).

evodevosys, 2021-06-25:
Thanks for figuring that out - I hadn't paid attention to keyboard vs. mouse.  So for now, I'll just use the mouse to navigate that section.

closed 2026-09-29 by prusaslicer-bot[bot]