prusa3d/PrusaSlicer #6647: Option to select print surface to accomodate for sheet thickness

jrsall92, 2021-06-23:
### Version
2.3.1

### Operating system type + version
MacOS

### 3D printer brand / version + firmware version (if known)
Mini+ 4.3.1

### Behavior
* I want to select the print sheet from the slicer, to accommodate for different thicknesses
* Search for "sheet" or "surface" in the settings
* The option to select among different surfaces
* No results
  *
[image]
[image]

_Is this a new feature request?_ Yes
I know that in recent Prusa FWs there is the option to set up different sheet profiles in the printer, but I'd like to have that functionality in the slicer. One way to go about it is to have the slicer accept the z offset and the print sheet used for the z offset calibration and from there calculate the difference in the different sheets.

ssill2, 2021-06-23:
While I agree, I think the inherent danger is taking .3mf or gcode from someone with different settings than you and crashing the nozzle into the plate.

I would prefer to have the printer prompt you for which sheet to use at the start of your print.  or at least make that prompt an option.
I definitely understand not using the the zOffset defined in the printer profile from the .3mf.

jrsall92, 2021-06-23:
I didn't think of that as I never do that, I always slice my prints. Having both, where you can enable the question in the printer and/or have it available in the slicer, would give the most flexibility I think.

ssill2, 2021-06-23:
yes, but if you open someone else's .3mf and it has a zOffset defined that's different from yours and you slice that to gcode, you could get into a case where it's radically different from yours.  I have two prusa built MK3's and the zoffsets for them are very different.  I often slice stuff for both printers.  It could easily get me into a situation where even with reslicing could get me in trouble if the gcode included the zoffset...

jrsall92, 2021-06-23:
Sorry, I meant that the printer's offset from the sheet profile should overwrite the slicer's offset, if there are profiles stored on the printer or if the option to select a sheet is enabled

ssill2, 2021-06-23:
well I'm not sure what others think but if you've taken the time to define profiles in the printer(I do this very often for tpu, petg pla for smooth and textured sheets) then it doesn't seem to make sense to try to force them in the slicer.  As I said, I have two MK3S+ printers assembled by prusa.  the offsets between them for the same sheet are very different.  When you sliced you'd have to make sure you sliced for THAT printer.  I'd love for the printer sheet profiles to be associated with a filament type like FLEX or PET.  This way if you start a print that you preheated for FLEX but the currently selected sheet profile is Tpetg then it could prompt you somehow.  It's easy to forget to change your selected sheet profile between prints.

rtyr, 2021-06-23:
I don't think this is a task for PrusaSlicer. It is also not possible to calculate it automatically, even the same sheet type can have different revisions/thickness, probes don't have 100% identical sensitivity, there are personal preferences about 1st layer, etc.. 

What you can likely do is to create separate printer profiles with M851 Z-x.xx command (liveZ value) in the start g-code or ask firmware team to introduce some g-code command to trigger sheet selection on the printer side.

jrsall92, 2021-06-23:
If I know the thickness of each sheet, let's say by using some calibers, I don't see why I can't have that in my slicer, apart from downloading ready made gcode, which imo is bad.

ssill2, 2021-06-23:
I think rtyr is right.  copy the printe profile as "MK3_mysheet" and put some custom gcode in.  Select that as your printer when you print and in theory it should do what you want.

jrsall92, 2021-06-23:
I can also edit the gcode file -.-

The point is to have a new, easy to use, feature about it, not find a round about way to do it.

ssill2, 2021-06-23:
having a printer profile with some custom start g-code is about as easy as you can get IMHO.

ssill2, 2021-06-23:
So just take the built in mk3s+ profile and save as a new name
[image]

Then add the custom gcode to that, the first line is the one rtyr pointed out.
[image]

From then on when printing just pick that printer as your printer along with your printer settings and filament.

ssill2, 2021-06-23:
Either way, either when you slice it, or when you print(by selecting sheet profile), you'll have to remember to change the printer or the sheet.  The printer simply has no way to know if you put a textured sheet, smooth sheet or satin sheet on or which sheet in fact since each can have variance.

rtyr, 2021-06-23:
I would rather have a possibility to print the g-code on any sheet I select on the printer side, instead of being limited by hardcoded liveZ value...

ssill2, 2021-06-23:
yes, and I find I tend to live adjust z for some filaments and not others.  Not all my PETG behaves the same way for instance.   I'd be happy with better and more naming choices for the steel sheet profiles in the firmware.

ruedli, 2021-07-01:
I feel this should not be chosen (and fixed) at slicing time. I prefer to decide on which sheet to use at printing time. 

Settings along live-adjust-z tend to change over time, e.g. when you have worked on the frame and changed to distance of the selector, so going the route to include that in the print file settings is even worse.

ssill2, 2021-07-01:
Set up steel sheet profiles for this.

ssill2, 2021-07-01:
specifically
https://help.prusa3d.com/en/article/steel-sheet-profiles_1955/

rtyr, 2021-07-01:
I believe we can close this. This is not a task for PrusaSlicer.

closed 2021-07-01 by rtyr