prusa3d/PrusaSlicer #6648: PrusaSlicer looses print profiles

marcocipriani01, 2021-06-23:
### Version
PrusaSlicer 2.3.1

### Operating system type + version
Windows 10 version 21H1

### 3D printer brand / version + firmware version (if known)
Anycubic Mega S with Klipper

### Behavior
Some time ago I switched from Cura to PrusaSlicer, and I made some print profiles for my printer starting from the system preset for the Anycubic Mega S (and saved all of them). Later on, I've noticed that the default Anycubic "Print Settings" had disappeared from the dropdown, but I thought it wasn't a big deal because I had saved some custom ones. Today, when I opened PrusaSlicer, I had zero "Print settings" in the dropdown. Only the `- default -` profile. However, all my "Filament" and "Printer Settings" are still there.

Honestly, I have no clue about what it could be, as I haven't messed around with the config files nor updated PrusaSlicer. I have no way of reproducing the issue. Hopefully someone has some thoughts about it...
Kind regards,
Marco Cipriani

rtyr, 2021-06-23:
Can you please zip your config folder and post it here? (Help -> Show configuration folder).

marcocipriani01, 2021-06-23:
Here it is: [PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/6703319/PrusaSlicer.zip)
Strangely enough, I've just discovered that all my "Print Settings" INI files are in the configuration folder. For some reason, they are not being loaded.

rtyr, 2021-06-23:
It is caused by compatible condition set in your print profiles (profiles are visible only for printers with 0.4mm nozzle). Remove or modify selected part.

[image]

marcocipriani01, 2021-06-23:
Oh so that condition was the problem. I did indeed change the nozzle diameter to 0.5 some days ago. I've messed around with the config files and I've managed to also restore the system presets. Thanks for your help!

closed 2021-06-23 by marcocipriani01