prusa3d/PrusaSlicer #6650: show estimated print time on ruler is not working

pixsrv, 2021-06-23:
### Version
Wersja 2.3.1+win64

### Operating system type + version
win 10

### Behavior
show estimated print time is not working
the checkbox can be ticked and unticked but the time is not presented

rtyr, 2021-06-24:
You have to hover mouse cursor over the ruler to see the estimated time.

[image]

pixsrv, 2021-06-24:
yyy... i feel a bit stupid :)
there is an issue anyway
checkboxes are working in a strange way
no matter which you click they check in some order but not in a way you expect
(but it might be other issue to report)

rtyr, 2021-06-24:
I think it's intended behavior. It is not classic checkbox.

pixsrv, 2021-06-24:
oh... this seems to be a magic-knowledge-needed ;)

first
i think when there is checkbox which turns on a tooltip (for print time) it should be mentioned that it will be just a tooltip
especially when it's next to the regular one which turns on something visible instantly (but not on hover)

second
if a checkbox is not working as regular-checkbox it should be also mentioned (on a tooltip on this checkbox?)

2 magic things on one feature... it deserves to take care of it i would say

i know i can used to it but... well it just could follow the regular control behavior

rtyr, 2021-06-24:
@YuSanka will look into second thing, it seems indeed somewhat weird.

rtyr, 2021-06-24:
Fixed on master. Closing.

closed 2021-06-24 by rtyr