prusa3d/PrusaSlicer #6651: PrusaSlicer does not show changes  at layer when .3MF when imported

gorobey, 2021-06-25:
### Version
PrusaSlicer-2.3.1+win64-202104161404

### Operating system type + version
Windows 10 (build 19043), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3s+ / MMU2 / 3.10.0 Firmware

### Behavior
when i import a .3mf project with change color, this is not show in layer "timeline" (vertical progress bar at the right).
but if i move at the specific layer, icon for remove changes appair

Try with this file:
[gravity_spoolholder_twocolor_left_1x.zip](https://github.com/prusa3d/PrusaSlicer/files/6714736/gravity_spoolholder_twocolor_left_1x.zip)

Source files:
https://www.prusaprinters.org/prints/15665-gravity-spoolholder-for-mmu2s-3mf-03mm-pla-two-col/files

gorobey, 2021-06-25:
ok, isn't a bug in import, "it's just" an inconsistency in UI.

rtyr, 2021-09-25:
> when i import a .3mf project with change color, this is not show in layer "timeline" (vertical progress bar at the right)

It is displayed correctly, the other color is just very similar. It may be difficult to see it on some monitors. Color can be changed (see screenshot) and then saved to 3MF.

[image]
[image]

Closing.

closed 2021-09-25 by rtyr