prusa3d/PrusaSlicer #6652: Idea, new Feature: Convert Volume into infill

Kopplog, 2021-06-25:
As explained in the headline. 
This would bring lot of possibilities (eg. I need) 
Thx

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-06-28:
Please provide more detailed description, examples, screenshots, etc.

Kopplog, 2021-06-28:
Hi
I'm Industrialdesigner.
Recently Im prototyping a lot whilst one parts I am working on resembles an aluminum extrusion part-to-be in the future. Now, such extrusion parts by nature won't have closed caps on their ends, as they're more to think as "possibly endless" rods done by that manufacturing process. Still, a plastic part (from FDM) in any case, won't have material properties just like eloxed aluminium...
So, what I was looking on the PRUSASLICER was a tool, which gives me an infill-structure to I add to part part-to-be-printed. (This infill-volume) would sit within the technical struts of the parts - not closed up from either sides, the heatbed nor the top, BUT would stiffen the whole to the point so I could use it for technical test runs.
(To model in fills like that with my solidmodeller would be a too much work):
I could think of modeling/exporting a simple volume in this case and being able to convert that in the PRUSASLICER would bring a whole new useful, professional possibilities.

Thanks for reading, chris

wavexx, 2021-06-29:
So if I understand correctly, you have a tube which is modeled as such (with a hole), but you'd like PrusaSlicer to fill the hole?

rtyr, 2021-06-29:
> 
> 
> So if I understand correctly, you have a tube which is modeled as such (with a hole), but you'd like PrusaSlicer to fill the hole?

It should be already possible on current master.

[image]

wavexx, 2021-06-29:
@rtyr I guess this only works for holes in each slice. So it work would for a vertical tube, not an horizontal one.
(didn't even know this existed ;))

rtyr, 2021-06-29:
> 
> 
> So it work would for a vertical tube, not an horizontal one.

Correct.

Kopplog, 2021-06-29:
> So if I understand correctly, you have a tube which is modeled as such (with a hole), but you'd like PrusaSlicer to fill the hole?

Very correct, that is.
But a vertical standing, tubus - no caps wanted on both ends.

bubnikv, 2021-06-30:
https://3dlabprint.com/
https://www.thingiverse.com/tag:3dlabprint
they create airplane models by defining overlapping geometetries. One has to use the "Even / Odd" slicing mode, which is new in the upcoming PrusaSlicer 2.4. The trick they do is to define a single perimeter, no infill and they let slicer to offset each of the surface inside by half the perimeter line width. It is not exactly what you are asking for I suppose, but it works for them.

closed 2026-09-29 by prusaslicer-bot[bot]