prusa3d/PrusaSlicer #6653: PrusaSlicer: "Filament Runout"-Error

omexlu, 2021-06-25:
Hello,

I have a strange problem with my Artillery Genius in connection with the PrusaSlicer, as soon as I want to start a print it shows me the error "Filament Runout", but this only happens when I have sliced with the PrusaSlicer (so I assume an error here, or a command that does not fit in the start code).

Attached is the G-Code file that was sliced with PrusaSlicer:

https://easyupload.io/ojbql3?fbclid=IwAR2sn1Xi7Wv354WgANiavz1fJZfGZq7e9Nt8flA1zgAhdsxXZhV3a_UmOCE

PS. Just wanted to try the first print with PrusaSlicer then this error 😛

rtyr, 2021-06-25:
I suppose your printer is not stock. Provided g-code is also not sliced with stock profile. The issue is likely caused by the combination of your start g-code (you can compare with your Cura start g-code) and your hardware/firmware (bltouch connected to the fsensor connector?).

rtyr, 2021-06-25:
@SzabolcsHornyak May help you with your modified printer. He included BLTOUCH version of the Genius profile, so he may point you to the correct firmware etc.

omexlu, 2021-06-25:
All Is Stock, no modifications, no BL-TOUCH, only thing i changes in the prusa Profil Is to restore the mesh thats all.

Only the firmwares on the Printer are are Last version.
But i dont think this Is the Problem, cura Can handle that without a Problem :/

Dont understand why i am getting this error here only with pursaslicer.

rtyr, 2021-06-25:
Ok, I saw BLT_ in the filename and also M420 command, so it seemed like you are using bltouch.

Maybe also provide some Cura g-code so we can compare.

rtyr, 2021-06-25:
I wonder why are you restoring the mesh if your printer is stock? (no MBL)

omexlu, 2021-06-25:
The file is the preparation for a BLTouch :D wanted to print the bracket for it.

Marlin and BigTreeTech each has the latest version installed (with Mesh-Level).

Here are the codes from Cura:

START-CODE:
```
G28 ; home all axes
 M420 S1 ; take MBL-settings
 M117 Purge extruder
 G92 E0 ; reset extruder
 G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
 G1 X2 Y20 Z0.3 F5000.0 ; move to start-line position
 G1 X2 Y200.0 Z0.3 F1500.0 E15 ; draw 1st line
 G1 X2 Y200.0 Z0.4 F5000.0 ; move to side a little
 G1 X2 Y20 Z0.4 F1500.0 E30 ; draw 2nd line
 G92 E0 ; reset extruder
 G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
```

END-CODE:
```
G91; relative positioning
 G1 Z1.0 F3000 ; move z up little to prevent scratching of print
 G90; absolute positioning
 G1 X0 Y200 F1000 ; prepare for part removal
 M104 S0; turn off extruder
 M140 S0 ; turn off bed
 G1 X0 Y300 F1000 ; prepare for part removal
 M84 ; disable motors
 M106 S0 ; turn off fan
```

There not more,
Thanks in advance.

rtyr, 2021-06-25:
I am confused. As far as I know, stock Artillery Genius doesn't have any mesh bed leveling, only manual leveling with mechanical knobs. Yet you have FW with MBL enabled and for some reason recalling mesh with M420 command. I guess you have already FW for BLTOUCH, but BLT is not physically installed yet, which may be the issue. Filament runout is handled by firmware based on the triggered (or not triggered) sensor. I am not familiar with this printer, but as far as I know, bltouch is often connected (and configured in FW) to the same pins as former filament sensor pins...

Anyway, I don't see any issue on the PrusaSlicer side. The g-code from Cura looks quite similar, no big differences. I noticed Z movement before G28 in PS g-code, which is a little bit weird.

omexlu, 2021-06-26:
I have changed the firmware so MBL is enabled :) the filament sensor is attached to the display, bltouch and filament sensor are not on same port, normally this don't have any to do with the slicer ythats because I don't understand why this happens 🙈

omexlu, 2021-06-27:
There must be a bug either in PrusaSlicer or in Genius Profil, all other Slicer works without trouble.

rtyr, 2021-06-27:
Bundled profile is meant for stock printer (with stock fw, etc.). The profile is ok as long as it works fine with unmodified and fully functional printer.

I am afraid you will have to experiment/troubleshoot what start gcode will work with your fw/hw. It hopefully shouldn't be so hard to figure it out by editing the final gcode, connecting via terminal to see where exactly it happens, etc. 

PrusaSlicer just emits what is configured in the profile.

omexlu, 2021-06-27:
I will give a try with cura startcode and hopefully It Works :/ otherwise i need to return to cura.

Its strange because cura Can handle the minimal changes on firmwares and TFT firmwares :/

rtyr, 2021-06-27:
I think the fix will be very simple, you just need to find out where exactly is the problem.

omexlu, 2021-06-27:
Yes but how 🙈 im not very similuar with debugging.

SzabolcsHornyak, 2021-06-27:
It sounds a weird problem. I never heard a similar one.

So when you start printing "runout sensor error" is appearing in the LCD screen?
The runout sensor is connecting to the LCD board. You could deactivate the rounout sensor in the LCD menu.
Please turn off an try print with Prusa and Cura profile. Then turn on and try again both.

omexlu, 2021-06-27:
Yes it's connected to the TFT, yes correct this is show on the display.

I will try this tomorrow, I think then I would work but after returning on it will return to the same result, but let's give it a try tomorrow 😀

omexlu, 2021-06-28:
So I tried to take the Gcode from Cura, here the same, so is more generally with some problem with PrusaSlicer and the Genius :/

closed 2026-09-29 by prusaslicer-bot[bot]