prusa3d/PrusaSlicer #6654: Essential supports are missing

ZbynekT, 2021-06-25:
### Version
2.3.1
### Operating system type + version
Windows 8.1

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
Essential supports are missing with auto-generate feature and I didn't find a way to generate them via settings in export mode nor using custom suports.

#### Project File (.3MF) where problem occurs
[image]
[image]

[BatteryPack_Top.zip](https://github.com/prusa3d/PrusaSlicer/files/6718271/BatteryPack_Top.zip)

rtyr, 2021-06-25:
Disable "Don't support bridges" option.

[image]

ZbynekT, 2021-06-26:
Thanks! I've missed that one.
Still, even with bridging there are lines 80mm long hanging in the air which could be detected as "unstable" and suggest some options.
Cheers.

rtyr, 2021-06-28:
> Still, even with bridging there are lines 80mm long hanging in the air which could be detected as "unstable" and suggest some options.

There is already an issue related to that https://github.com/prusa3d/PrusaSlicer/issues/3859.

Closing.

closed 2021-06-28 by rtyr