prusa3d/PrusaSlicer #6655: wxWidgets3.0 again wxGetApp not declared in this scope

clytle374, 2021-06-26:
Gentoo Linux with current git source, problem began a couple weeks ago. Still running wxwidgets 3.0, hopefully 3.2 will be done soon.  If you are still trying to support 3.0 here is my build failure 
version_2.4.0-alpha0-656-g9b70efc46
`[ 74%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GUI_Utils.cpp.o
/home/cory/sources/prusa/src/slic3r/GUI/GUI_Utils.cpp: In function ‘bool Slic3r::GUI::check_dark_mode()’:
/home/cory/sources/prusa/src/slic3r/GUI/GUI_Utils.cpp:168:27: error: ‘wxGetApp’ was not declared in this scope; did you mean ‘wxApp’?
  168 |     const unsigned luma = wxGetApp().get_colour_approx_luma(wxSystemSettings::GetColour(wxSYS_COLOUR_WINDOW));
      |                           ^~~~~~~~
      |                           wxApp
[ 74%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/MainFrame.cpp.o
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:668: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GUI_Utils.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1286: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
`
Thanks, Cory

combolek, 2021-06-26:
I think the problem is that `#include "GUI_App.hpp"` is under `#ifdef _WIN32` but the `wxGetApp()` is called on all platforms (with wxWidgets 3.0 at least). But I don't know which way is correct. As a workaround try moving the include outside of the ifdef, e.g.
```diff
diff --git a/src/slic3r/GUI/GUI_Utils.cpp b/src/slic3r/GUI/GUI_Utils.cpp
index 78548e111..9286a2f2c 100644
--- a/src/slic3r/GUI/GUI_Utils.cpp
+++ b/src/slic3r/GUI/GUI_Utils.cpp
@@ -4,9 +4,9 @@
 #include <boost/lexical_cast.hpp>
 #include <boost/format.hpp>
 
+#include "GUI_App.hpp"
 #ifdef _WIN32
 #include <Windows.h>
-#include "GUI_App.hpp"
 #include "libslic3r/AppConfig.hpp"
 #include <wx/msw/registry.h>
 #endif
```

clytle374, 2021-06-27:
Thanks, That builds and runs
Cory

bubnikv, 2021-06-28:
Thanks, fixed with e6295bbfe813f9331354ccbfbe2aae4aa0b79aab

closed 2021-06-28 by bubnikv