prusa3d/PrusaSlicer #6657: Exported Gcode produces malformed print object

DrJuJu, 2021-06-26:
### Version
Version 2.3.1+win64

### Operating system type + version
Windows  10 Pro Version 20H2

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk2 
Firmware 3.2.3

### Behavior
A simple model comprising two square tubes, with supports and fixing holes. All square tubes sides are 24 mm, with a central 20 mm, square 'hole'. 

This should result in tube sides all being square and 2 mm thick, but the lower section vertical sides are thinner and leaning in, resulting in a lip where the sides meet the top horizontal section.

This has only started happening since I upgraded to 2.3.1 earlier versions did not result in printed model exhibiting this behaviour.

_Is this a new feature request?_ no

#### Project File (.3MF) where problem occurs

[Print_Error](https://user-images.githubusercontent.com/13620557/123509108-9388e880-d66b-11eb-962c-dbfecaa5d02a.jpg)
[Print_Error_Files.zip](https://github.com/prusa3d/PrusaSlicer/files/6719898/Print_Error_Files.zip)

DrJuJu, 2021-07-01:
The 'upper' part is perfectly square\vertical etc. it is just this lower part, and only happens with this version of the slicer.

I've also checked all the mechanical alignments & all is good....

rtyr, 2021-07-21:
I am afraid it's about physics, it will be affected by filament, cooling, ambient temperature, infill percentage, ...

Out of curiosity, I tried to print it with 3 different slicers.
[image]

See this comment for more information (last 2 paragraphs for this particular issue). https://github.com/prusa3d/PrusaSlicer/issues/1719#issuecomment-458447660.

> ...
> ...
> Lastly, it is the physics as others noted. Plastics, if extruded first from pellets into a filament, and then from filament into a very thin rectangular extrusion, will behave neither as a liquid, nor as a solid. The stretching of the plastic will align the long molecular chains of the polymer, introducing internal stresses to the extrusion. This internal stress will pull the extrusion together if not cooled quickly enough. Indeed, as many of you found out, in a cool basement the effects are more pronounced, also on my MK2 the 3D Benchy has the line more pronounced at the side away from the cooling fan.
> 
> So, it is "just" physics, and it will be very difficult to compensate for the internal stresses and cooling effects of the filament IMHO, one of the reasons being the viscoelastic behavior of the molten plastics, and the dependence of the plastic behavior on its composition, temperature, hydrolisis of the polymer molecular chains if the filament is not 100% dry (polyesters - PLA & PET are sensitive to hydrolisis, hydrolised filament contains shorter polymer chains, thus being less viscous), effectivity of the cooling, reflections of the cooling air from the already printed objects etc.
> 
> 
>

bubnikv, 2021-09-22:
No response, closing.

closed 2021-09-22 by bubnikv