prusa3d/PrusaSlicer #6658: Feature Request: Toggle visibility for Model in bed shape.

tacohaha, 2021-06-27:
Feature Request:
In printer settings, you can add a model to the bed shape. This for instance allows you to add a model of your printer so you can more easily see the scale of your print. Unfortunately, because you have to add the model to a printer profile, the only way to not see the model you've imported is to either remove the model or switch to another profile without the model. I would suggest adding a checkbox to toggle the model on and off.
[image]

rtyr, 2021-06-28:
So you have separate printer profile with custom bed model (printer model). And you want some simple way to toggle this model on/off (keyboard shortcut, checkbox,...)? Do I understand it correctly?

[image]

tacohaha, 2021-06-28:
Yes exactly.

closed 2026-09-29 by prusaslicer-bot[bot]