prusa3d/PrusaSlicer #6660: Disable solid (purple) infill along chamfer and angles (feate request)

GlennovitS, 2021-06-29:
### Version
2.3.1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa Mini+ (firmware 4.3.1)

### Behavior
When slicing a simple box with a chamfer on the bottom edges, PrusaSlicer adds purple infill along the inside of the perimeters. It should be to help with strength i guess, but at fx. a 45 degree chamfer I might want the 'lesser strength' and avoid the time and material for that 'solid infill'. 
Would LOVE a toggle to enable/disable this solid infill. Or some kind of adjust of at which angles it appears or something like that. 

Main reason for me to get rid of this solid infill is the insane amount of very small and fast movements. It can't be good for the printer to go through that, at least when I dont want it to do that anyways. And a second reason could be to reduce time and material used.

It shows up with default settings. Can sometimes be removed when selecting fx. 3 perimeters and 3 top/bottom layers. But I dont want equal amount of perimeters and top/bottom layers. I often use 2-3 perimeters and 4-5 bottom/top layers. 

_Is this a new feature request?_ Yes. Based both on my own experience AND a lot of googling where I could see that I'm not the only one that would love to get rid of this kind of solid infill. Would rather have the 'Cura style' where the normal infill just goes out to the inner perimeter (if that makes sense).

#### Project File (.3MF) where problem occurs
[image]
[Chamfer test with purple infill project.zip](https://github.com/prusa3d/PrusaSlicer/files/6731297/Chamfer.test.with.purple.infill.project.zip)

neophyl, 2021-06-29:
Isn't this a duplicate of #1054 ?

GlennovitS, 2021-06-29:
Yes. Sorry hadn't seen that thread/issue. Was told from a Prusa support chat that I should send a suggestion in here, so I did. Hopefully they will fix this if they actually read these issues. Can see that the issue you are linking to is very old! :-O

rtyr, 2021-06-29:
Don't worry, we read it. We will likely add more options for "ensure wall thickness". Closing as a duplicate.

closed 2021-06-29 by rtyr

GlennovitS, 2021-06-29:
> Don't worry, we read it. We will likely add more options for "ensure wall thickness". Closing as a duplicate.

Happy to hear. Looking forward for updates on this :-) and sorry for the duplicate post