prusa3d/PrusaSlicer #6671: 2.3.2-Beta Lost all settings

rubrettdev, 2021-07-03:
### Version
2.3.2-Beta

### Operating system type + version
macOS 12.0 Monterey

### 3D printer brand / version + firmware version (if known)
Sapphire Plus - Klipper (latest)

### Behavior
After upgrading from 2.3.1 to 2.3.2-Beta I've lost all settings and if I try to recover my configuration Snapshots I have an empty withe window.

[image]

[image]

How can I recover all my profiles that took months to tune?

rtyr, 2021-07-03:
Don't worry, 2.3.2-beta just uses different data directory than final release. Your profiles will be still there, if you start PS2.3.1 (or 2.3.2 final, which will be likely released next week). 

You can copy your profiles from 2.3.1 to 2.3.2-beta. Export your settings from 2.3.1 via File -> Export -> Export Configuration Bundle... and then import them to 2.3.2-beta using File -> Import -> Import Configuration Bundle...

rtyr, 2021-07-05:
PS2.3.2-rc was just released, it saves its profiles into PrusaSlicer directory, which is the same directory where the previous releases of PrusaSlicer (including 2.3.1) keep their profiles.

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.2-rc

Closing.

closed 2021-07-05 by rtyr