prusa3d/PrusaSlicer #6672: Split to objects/parts not working anymore in 2.3.2-rc2

BuysDB, 2021-07-04:
### Version
2.3.2-beta+linux-x64

### Operating system type + version
Kubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2

### Behavior
Split to parts/objects option is greyed out, for an object which could be split in version 2.3.0-rc2
* Steps needed to reproduce the problem
Import a STL file, rightclick, then `Split` is greyed out

* Expected: split not being greyed out  (In version 2.3.0-rc2 it works as expected)

rtyr, 2021-07-04:
You are likely in the simple mode, switch to advanced or expert.

BuysDB, 2021-07-04:
> You are likely in the simple mode, switch to advanced or expert.

I had no idea that this setting would influence this behavior. Switching to expert fixes the issue, thanks!

closed 2021-07-04 by BuysDB