prusa3d/PrusaSlicer #6673: Using Absolute E distances not working

kylewilk567, 2021-07-04:
### Version
2.3.1

### Operating system type + version
Windows 10
OS Build: 19042.1052

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 Original with MK3S upgrade

### Behavior
When you uncheck the box "Use relative E distances" under Printer Settings -> General (Expert setting), it generates gcode that doesn't completely use absolute E distances. It resets the coordinates each layer or so. See the text file link below for the gcode, it shows it changes to absolute coordinates (M82 command) but the E coordinates reset back to around 2.xx each subsequent layer. While doing a test print with this gcode, it also only prints the first layer.

[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Prusa_GcodeAbsoluteExtrusionError.txt](https://github.com/prusa3d/PrusaSlicer/files/6760446/Prusa_GcodeAbsoluteExtrusionError.txt)

rtyr, 2021-07-04:
You probably forgot to remove G92 E0 command from the "before layer change g-code" custom g-code section.

kylewilk567, 2021-07-04:
Oh I see that now. I didn't realize it did that. Thank you!

closed 2021-07-04 by rtyr