prusa3d/PrusaSlicer #6676: Cant rotate view when creating supports for SLA models

assafwei, 2021-07-05:
### Version
_2.3.1_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 10 pro and home versions
### SL1

### Behavior
* In SLA slicing (maybe happens in FDM too, didnt try), when building supports for example, when trying to rotate the view, if the mouse pointer is on the model, I cannot rotate the screen, only if I click on a space without any model (or supports) I am able to rotate the model.
* Import a model, put curson on model and try to rotate during support editing.
  * _If this is a command-line slicing issue, include the options used_
* Rotation at any point the cursor is on
*No rotation of the screen when mouse cursor is on model

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

FidelCapo, 2021-07-07:
You can use Ctrl with mouse move for camera rotate when some gizmo is open.
Did that help?

bubnikv, 2021-09-22:
No answer, closing.

closed 2021-09-22 by bubnikv