prusa3d/PrusaSlicer #6677: [Linux] Certain STLs will crash SuperSlicer when opening layer view on Ubuntu 21.04 with AMD hardware

ChaosBlades, 2021-07-05:
### Version
PrusaSlicer 2.3.1+linux-X64

### Operating system type + version
Ubuntu 21.04
Pop!_OS 21.04

### Behavior
1. Open PrusaSlicer (has no .config / new install)
2. Click Cancel on wizard
3. Add Crocodile Clip or All In One STL below
4. Click "Slice now"
5. Application Crashes once processing finishes to calculate the color filters

### Project File (.3MF) where problem occurs
[Crocodile_clip_v1.2.zip](https://github.com/prusa3d/PrusaSlicer/files/6766394/Crocodile_clip_v1.2.zip)
[AIl_In_One_3D_printer_test_micro.zip](https://github.com/prusa3d/PrusaSlicer/files/6766395/AIl_In_One_3D_printer_test_micro.zip)

### Additional Details
Laptop with Intel / Nvidia
Ubuntu 20.04 = Works
Ubuntu 21.04 = Works

Laptop with AMD CPU/GPU
Ubuntu 20.04 = Works
Ubuntu 21.04 = Crash

My Desktop that this is also crashing on is also 21.04 with AMD CPU/GPU. 

Here is the original issue I made with SuperSlicer. This issue is also reproduce-able on PrusaSlicer.
https://github.com/supermerill/SuperSlicer/issues/1389

bubnikv, 2021-07-06:
> supermerill#1389 <https://github.com/supermerill/SuperSlicer/issues/1389>

It is always good to attach a 3mf, which contains all the parameters to
slice with.

ChaosBlades, 2021-07-06:
It is 100% stock. No .config folder / fresh install. No settings are configured or applied.

1. Open PrusaSlicer for the first time
2. Close wizard with cancel button
3. add STL
4. Slice
5. Watch it crash (if Ubuntu 21.04 w/ AMD Hardware)

P.S. Cura works without issue.

hakimio, 2021-07-06:
Most likely related: https://github.com/prusa3d/PrusaSlicer/issues/6662

ChaosBlades, 2021-07-06:
First one occurs on 20.04 which my issue does not. 
Second one was occurring/fixed in "5.7-rc1 - 5.7 - 5.8-rc5+, fixed in 5.8.0, 5.7.13, 5.4.56". I am on Kernel: 5.11.0-7620-generic

hakimio, 2021-07-06:
@ChaosBlades If you check the comment I linked on that kernel thread, you'll see that Christopher Snowhill is experiencing the bug with 5.10.4.
Anyway, do you have any errors on `dmesg` when it crashes?

ChaosBlades, 2021-07-06:
Yes I do.
`traps: slic3r_main[68698] general protection fault ip:7f62c9ba7a3d sp:7ffe5501f690 error:0 in radeonsi_dri.so[7f62c9116000+f00000]`

hakimio, 2021-07-06:
Well, the error is the same. So, it might still be related even you can't reproduce on 20.04
Maybe it's worth asking **Duncan** on that kernel thread if this issue has anything to do to the one he solved?

senseisimple, 2021-07-09:
I'm having the same issue on Manjaro Arch, gnome 3/gtk3/qt5 , AMD Vega10 gpu.

Same trigger (as soon as done slicing right when feature colors are applied) and symptom (crash) as noted in both github issues for  PrusaSlic3r and SuperSlicer starting at version 2.3.0 to current with existing configs and new configs.

prusaslicer 2.20 (and superslicer 2.2.53) both work flawlessly.

base system specs:
```
Linux 5.11.4-1-rt11-MANJARO #1 SMP Fri May 21 00:10:17 CEST 2021 x86_64 GNU/Linux

#qtdiag
Qt 5.15.2 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 11.1.0) on "xcb"
Platforms available : 
  gtk2,gtk3,qt5ct,snap,flatpak,xdgdesktopportal
Available physical devices:
  API version 1.2.178, vendor 0x1002, device 0x6861, Radeon Pro WX 9100, type 2, driver version 2.0.188
Vulkan:
  Min Texture Size: 1
  Max Texture Size: 16384
  Max Color Attachments: 8
  Frames in Flight: 2
  Uniform Buffer Alignment: 16
  Supported MSAA sample counts: 1,2,4,8

#lspci (graphics)
01:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Vega 10 PCIe Bridge
02:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Vega 10 PCIe Bridge
03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 XT [Radeon PRO WX 9100]
03:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 HDMI Audio [Radeon Vega 56/64]

#lsmod (gpu)
amdgpu               6729728  98
gpu_sched              45056  1 amdgpu
i2c_algo_bit           16384  1 amdgpu
drm_ttm_helper         16384  1 amdgpu
ttm                    86016  2 amdgpu,drm_ttm_helper
drm_kms_helper        282624  1 amdgpu
drm                   569344  22 gpu_sched,drm_kms_helper,amdgpu,drm_ttm_helper,ttm

#glxinfo (opengl/vulkan)
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 16384 MB
    Total available memory: 32752 MB
    Currently available dedicated video memory: 13884 MB
OpenGL vendor string: AMD
OpenGL renderer string: Radeon Pro WX 9100 (VEGA10, DRM 3.40.0, 5.11.4-1-rt11-MANJARO, LLVM 12.0.0)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.1.2
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: no-error
OpenGL core profile profile mask: core profile

#clinfo (opencl)
$ sudo clinfo
Number of platforms                               1
  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 21.1.2
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   Clover
Number of devices                                 1
  Device Name                                     Radeon Pro WX 9100 (VEGA10, DRM 3.40.0, 5.11.4-1-rt11-MANJARO, LLVM 12.0.0)
  Device Vendor                                   AMD
  Device Vendor ID                                0x1002
  Device Version                                  OpenCL 1.1 Mesa 21.1.2
  Device Numeric Version                          0x401000 (1.1.0)
  Driver Version                                  21.1.2
  Device OpenCL C Version                         OpenCL C 1.1 
[... 68 lines cut ...]
02:49:01 AM: Debug: window wxTreeCtrl(0x55ff68e1a800, ) lost focus even though it didn't have it
02:49:01 AM: Debug: window wxTreeCtrl(0x55ff68e1a800, ) lost focus even though it didn't have it

(prusa-slicer:1822507): Gtk-CRITICAL **: 02:49:24.814: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:1822507): Gtk-CRITICAL **: 02:49:24.814: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:1822507): Gtk-CRITICAL **: 02:49:24.902: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:1822507): Gtk-CRITICAL **: 02:49:24.902: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
Segmentation fault (core dumped)

```

dmesg output (relevant info):
```
#prusaslicer:
[1917053.404093] traps: slic3r_main[1809560] general protection fault ip:7f37ac588bdf sp:7ffd444fbcb0 error:0 in radeonsi_dri.so[7f37ac3ad000+fda000]
[1917053.404127] audit: type=1701 audit(1625814789.091:2419): auid=1000 uid=1000 gid=1000 ses=3 pid=1809560 comm="slic3r_main" exe="/tmp/.mount_PrusaS7GI8WE/usr/bin/bin/prusa-slicer" sig=11 res=1

#superslicer:
[1918672.551957] superslice:gl0[1819292]: segfault at 3560 ip 00007fbf1a7b0239 sp 00007fbeedff81a0 error 4 in radeonsi_dri.so[7fbf1a707000+fda000]
[1918672.551968] Code: 00 4c 89 fe eb cb 0f 1f 00 4c 89 fe eb 83 0f 1f 00 48 89 df e8 08 83 ff ff e9 c1 fd ff ff 0f 1f 00 48 85 d2 0f 84 45 ff ff ff <48> 8b 8a c8 00 00 00 48 85 c9 0f 84 2c ff ff ff 48 3b aa d0 00 00
[1918672.551996] audit: type=1701 audit(1625816408.219:2454): auid=1000 uid=1000 gid=1000 ses=3 pid=1819227 comm="superslice:gl0" exe="/usr/share/SuperSlicer/superslicer" sig=11 res=1
```

ChaosBlades, 2021-07-12:
Can confirm PrusaSlicer and SuperSlicer 2.2 both work without issue. Something in 2.3 is causing this issue.

senseisimple, 2021-07-13:
[broken-stl-example.zip](https://github.com/prusa3d/PrusaSlicer/files/6806175/broken-stl-example.zip)

Here is an example of an object that makes the sliced preview crash. The crash happens within 1mm (actually a 0.2mm layer) of a working cut. I have included .stl (SuperSlicer 2.3+ creates non-manifold cut when cutting sections), .obj (repaired manifold, same issues), .3mf, .gcode (sliced with SuperSlicer 2.2.53).

That is from a complete object that makes the slicer preview crash, I have isolated the 2.3*mm section that makes a reproducible crash, 
Mashroom (20)-25.49mm-top-2.30mm-OK (stl, obj) - slices/previews OK
Mashroom (20)-25.49mm-top-2.31mm-broken (stl, obj) - crashes on slice/preview

senseisimple, 2021-07-13:
It seems to have been narrowed down to the perimeter width and gapfill settings as detailed in https://github.com/supermerill/SuperSlicer/issues/1313#issuecomment-878803443 and further confirmed in the next comment.

Excerpts:

> I narrowed this down to the MedialAxis call when generating gapfill extrusions. This particular layer has some thin walls where only two perimeters and a big fat gapfill would fit.

> - Width and Flow: anything other than Default: 0.4mm (int/ext Perimeter 0) (0.2mm layer height) crashes
> - Infill: Top and/or Bottom: "Monotonic ( filled),", "Concentric (Filled) (any perimeter width) crashes
> - **Infill: Top and Bottom: Rectilinear (no fill) @ Width and Flow: Default: 0.4mm (int/ext Perimeter 0) (0.2mm layer height) NO CRASH but**

senseisimple, 2021-07-13:
Related (also AMD):
https://github.com/prusa3d/PrusaSlicer/issues/5745
https://github.com/prusa3d/PrusaSlicer/issues/6342
https://github.com/supermerill/SuperSlicer/issues/1057 (closed, no response)
https://github.com/prusa3d/PrusaSlicer/issues/6345

As noted in many comments, setting env variable `LIBGL_ALWAYS_SOFTWARE=true` alleviates the segfault.
PrusaSlicer >=2.3.1.* does not resolve.

ChaosBlades, 2021-09-18:
I recently got an update to Mesa 21.2.1 which appears to resolve this issue.

closed 2021-09-18 by ChaosBlades