prusa3d/PrusaSlicer #6681: OpenEXR 3 compatibility

worksasintended, 2021-07-07:
### Version
version_2.4.0-alpha0-689-g3fa78b52b

### Operating system type + version
ArchLinux and some other distros
### 3D printer brand / version + firmware version (if known)
unrelated

### Behavior
Cmake is unable to find the ilmbase lib which is due to the newer headers in OpenEXR3.
The file causing the issue is 
```cmake/modules/FindOpenVDB.cmake```
There is a patch floating around:
https://github.com/archlinux/svntogit-community/blob/packages/prusa-slicer/trunk/prusa-slicer-openexr3.patch

tamasmeszaros, 2021-07-15:
Hello! I've made some changes to how PS finds OpenVDB (which requires OpenEXR). In case of a distro build, one can now specify the path to OpenVDB find modules and bypass the modified FindOpenVDB bundled with PS. I've tried it with ArchLinux and it worked. The only thing you need to do is to specify the `OPENVDB_FIND_MODULE_PATH` cache var for CMake. 

In case of ArchLinux this was possible because the `openvdb` package installs the find modules in `/usr/lib/cmake/OpenVDB`
For Debian derivatives this might not be so easy, the find scrips are not part of the `libopenvdb-dev` package.

The posted patch would make FindOpenVDB.cmake work for OpenEXR3 but not for OpenEXR2, that is why I would not merge that into PS. Making the bundled FindOpenVDB.cmake ready for OpenEXR v2 and v3 at the same time would involve a lot of changes for this find module which I refuse to do. It was already a huge work to make it usable in the first place. It is also stated on OpenVDB docs that they might change the find modules to Config scripts which would solve all the problems here.

worksasintended, 2021-07-16:
Great, thank you!

closed 2021-07-16 by tamasmeszaros