prusa3d/PrusaSlicer #6682: The values to purge on unload won't allow more than 300

ssill2, 2021-07-07:
### Version
2.4.0 alpha   from latest master rev as of July 7 2021

### Operating system type + version
Windows 10 64 bit

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa MK3S+ with MMU2S

### Behavior
* _Describe the problem_
The problem I'm having with a particular two color-print(yellow and black), I'm getting obvious bleed-through into my yellow when the MMU changes from black back to yellow.  I've improved things by bumping the value for purge on unload to 300.  But I still see some obvious mix of the black in the yellow.  The slicer ui will not let me put in a value higher than 300.  I've been able to improve things by creating a part of the model I'm printing that's not connected to it but using that object as a sacrificial one and setting as wipe to infill.   But really the point of the wipe tower is to essentially be a sacrificial object and it seems I should just be able to increase the unload purge value to more than 300 until there's no visible blend.  I'd recommend increasing this to at least 500.   If someone can tell me which file this value is in I can try it locally first to see.

ssill2, 2021-07-07:
Ok I figured out which file it was in.  I'm doing some tests.

rtyr, 2021-07-08:
You can use advanced settings for higher value. In my experience, 300+300 is more than sufficient for simple mode.

[image]

ssill2, 2021-07-08:
ah ok, that works.   thank you!

closed 2021-07-08 by ssill2