prusa3d/PrusaSlicer #6683: Supports on build plate option still generating supports on top of parts of model

bstiq, 2021-07-07:
### Version
2.3.2-rc
Also tested on 2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3s
Prusa MK3s+
Prusa Mini+
All latest firmware

### Behavior
When generating supports, enabling option "supports on build plate only" doesn't have expected effect.
Prusaslicer generates supports on top of some parts of the model.

Used model:
[image]

Expected result:
[image]

Actual result:
[image]

Also tried support enforcers and blockers, mitigates problem but doesn't solve it.

#### Project File (.3MF) where problem occurs
[SK_V3_V6.zip](https://github.com/prusa3d/PrusaSlicer/files/6780067/SK_V3_V6.zip)

Regards,

jmd73, 2021-07-07:
@bstiq This is fixed in the 2.4.0 alpha branch.

[image]

bstiq, 2021-07-07:
Confirmed, thanks, You can close this ticket.
Another quick question if possible, is there some documentation on cases where slicing is extremely slow? 
15+ minutes on a ryzen 2700 (rather than the usual 20 seconds max on the same file in 2.3.2).

Thanks, 
Regards,

jmd73, 2021-07-07:
@bstiq Hello again!

I can't close this, since I am not an admin, just a normal user.
You can close this yourself, or wait for the Prusa people to close it.

To find earlier issues you can search closed and opened issues here in github.

bstiq, 2021-07-07:
Thanks @jmd73. I will try again tomorrow on another computer and if I still have the issue will create another issue.
Couldn't find anything in the issues list.

Regards,

closed 2021-07-07 by bstiq