prusa3d/PrusaSlicer #6686: Crash after creating large amount of objects after "Split to objects" operation

JohanAR, 2021-07-08:
### Version
Version 2.3.1+linux-x64-GTK3

### Operating system type + version
Xubuntu 21.04

### 3D printer brand / version + firmware version (if known)
Ender 3 v2, but not relevant I think

### Behavior
Having too many objects appears to crash PrusaSlicer. If I split the same mesh in Blender it turns into 6271 objects

To replicate:
1) Import an STL with large number of disjoint meshes in a single object
2) Split to objects
3) Select one of the objects

Alternatively:
1) Open the attached .3mf file
2) Select one of the objects 

Expected behaviour:
Does not crash when selecting object.
OR: Refuse to open file or split to objects when it results in too many objects.

#### Project File (.3MF) where problem occurs
[Fin_Supports_Rybonator_Dice_Files_1.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6783384/Fin_Supports_Rybonator_Dice_Files_1.3mf.zip)
Just remove the .zip extension, it's not double-compressed

bubnikv, 2021-12-08:
I am not getting crash with 2.4.0-beta3, however the application hangs at the end of slicing.

kubispe1, 2022-09-23:
Hi, I inspect this issue with release 2.5.0 -win64 (win10) with this steps:
- Open the attached .3mf file
- Select one of the objects  
- after that I try to merge them together
- and than I split it
- I also can slice it

It is really heavy job and slicing take a long time, but seem to me that it works, so I am closing it. If you still have a problem. Lets disscus it and in case I will reopen it. 

Thank you @JohanAR