prusa3d/PrusaSlicer #6691: Crash on some designs

Thijxx, 2021-07-09:
### Version
2.3.1 Linux x64

### Operating system type + version
Ubuntu 21

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Behavior
Some desings crash the slicer when generating Gcode. It's only the Prusa Slicer that cannot handle the designs. I found two at the moment. Debug, logging, command-line output, all empty. Try it yourself with the attached design. Unsliceable.

#### Project File (.3MF) where problem occurs
[GridCap-Small-52.zip](https://github.com/prusa3d/PrusaSlicer/files/6791978/GridCap-Small-52.zip)

rtyr, 2021-07-09:
Please provide your 3MF project file.

neophyl, 2021-07-09:
Slices fine for me on a Win10 64 bit install.  Default mk3 printer, generic PLA and a quality 0.2 print profile.

combolek, 2021-07-09:
I cannot reproduce it on Linux either. Both `PrusaSlicer-2.3.1+linux-x64-GTK3-202104161351.AppImage` and `2.4.0-alpha0` built from source work fine.