prusa3d/PrusaSlicer #6692: Slice button not appearing after infill change

Cralost, 2021-07-09:
I have version 2.3.1 (win64) and I find very annoying that after when I change model infill % I need to click somewhere else (activate different element) to Slice button appear.

fe60, 2021-07-27:
I can't reproduce the issue. I tried using the mouse, keyboard, used the drop down in the "Print settings" tab and the one above the object list and I switched to non-modal window to show the print settings. The Export/Slice button always gets updated as soon as the amount of infill has been changed.

Which version of Windows do you use (see PrusaSlicer Help=>System Info, the popup allows to copy its content)?
How exactly do you change the infill density (a video could show details you don't consider to be relevant)?

Cralost, 2021-07-27:
[image]

I was hitting also enter btn during recording. 

[image]

neophyl, 2021-07-27:
I cant reproduce on my system either.  Changing infill % updates the button for me.  Also on Windows 10 with the same build 19042  and 2.3.1

fe60, 2021-07-27:
When using the keyboard, I selected values from the drop-down list using the arrow `Up`/`Down` keys, but I never entered "own" values.

And you are right, pressing `Enter` after entering a value does not have any effect. You can observe this behavior for all editable drop-down lists, while `Enter` works as expected in other input fields.

bubnikv, 2021-08-11:
Fixed in PrusaSlicer 2.4.0-alpha1.

bubnikv, 2021-08-30:
Fixed with 83ab034f9a8769e2d2e8ce306bafa16682150234

closed 2021-08-30 by bubnikv