prusa3d/PrusaSlicer #6693: Brim too narrow

pbartko, 2021-07-09:
### Prusa Slicer version: Version: 2.3.1
### OS: Windows 10 (20H2)
### Printer: Prusa MK2S clone
### Problem: When using a 0.6mm nozzle and a Brim, Prusa Slicer creates brim extrusions that are too narrow

[image]
[Thread_Test_1_x1_Rev1 ASA.zip](https://github.com/prusa3d/PrusaSlicer/files/6794008/Thread_Test_1_x1_Rev1.ASA.zip)

neophyl, 2021-07-10:
Please read the tooltips.  You have first layer extrusion width set to 100%.  When set to a % its width is based on LAYER HEIGHT not nozzle size.  As your layer height is set to 0.2 then all your first layer extrusion widths are set to 0.2mm too.  

btw you don't have Brim enabled so why call the issue Brim ?  The thin extrusion widths apply to everything on the first layer.

rtyr, 2021-07-10:
Your 1st layer extrusion width is set to 0.2mm.

[image]

Closing.

closed 2021-07-10 by rtyr