prusa3d/PrusaSlicer #6694: "Wipe while retracting" even when disabled.

stevereno30, 2021-07-09:
PrusaSlicer Version 2.3.2 
Windows 11
Marlin 2.0.x bugfix (April 2021)

### Behavior
When "wipe while retracting" is unchecked, the slicer still adds wipe movements. Unknown if this happens in all printed parts, but it definitely happens in a single-walled test cube.
[Shape-Box_0.2mm_PETG__15m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/6794132/Shape-Box_0.2mm_PETG__15m.gcode.zip)

rtyr, 2021-07-10:
You have "wipe while retracting" enabled in filament overrides.

[image]

stevereno30, 2021-07-10:
That I do. I feel silly now! XD
Thanks!

closed 2021-07-10 by stevereno30