prusa3d/PrusaSlicer #6699: Incorrect slicing of .stl file

mgrider, 2021-07-11:
### Version / System Information
PrusaSlicer
Version:   2.3.1+x64
Build:     PrusaSlicer-2.3.1+x64-202104161356

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Big Sur Version 11.4 (Build 20F71)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   2.1 ATI-4.5.14
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon R9 M295X OpenGL Engine
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_blend_weighted_minmax
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
When adding the attached .stl file to PrusaSlicer, it incorrectly adds a notch near one of the corners of the model. The original modeling program (OpenSCAD) doesn't show this, nor does the MacOS preview of the .stl file.

Steps to reproduce:
1. unzip attached file
2. Open PrusaSlicer
3. Import STL file.
4. Observe upper-left corner of the model.
[image]

Here's what I expect the 3D model's corner to look like:
[image]

#### Project File (.3MF) where problem occurs
Here is a zip file containing both the .stl file, as well as a .3mf file showing the problem import.
[prusaSlicer_stl_import_error_examples.zip](https://github.com/prusa3d/PrusaSlicer/files/6796129/prusaSlicer_stl_import_error_examples.zip)

JohnOCFII, 2021-07-11:
There are errors in the model as shown but the slicer.  Sadly, OpenSCAD seems to be a particularly bad offender when it comes to creating damaged STLs.  I was going to suggest you run the model through the free Netfabb service, but it appears that STL repair is now part of Fusion 360 and not available separately.  https://service.netfabb.com
[image]

JohnOCFII, 2021-07-11:
The site: https://tools3d.azurewebsites.net can repair this file: 
[closet_bracket-2_repaired-3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6797032/closet_bracket-2_repaired-3mf.zip)

The attached file is a 3mf, renamed as a zip file to upload to Github.

rtyr, 2021-07-12:
The issue is still present in 2.3.2.

closed 2026-09-29 by prusaslicer-bot[bot]