prusa3d/PrusaSlicer #6700: Bridge flow ratio affecting all infill?

CZDanol, 2021-07-11:
So I've sliced my model and started printing. After some time, I've noticed weird extrusion artifacts and that the extruder is barely extruding. I've checked my model and everything seemed okay. Then I've noticed that when I was changing some parameters, I've accidentally set the bridge infill ratio to 0.095 instead of 0.95 that I itended. The thing is, the artifacts were visible on standard infill, not bridge infill. This makes it seem to me that the bridge infill ratio is incorrectly applied on non-bridge infill as well (that could also possibly explain my general slight underextrusion I've been experiencing).

[image]
[image]
The weird artefacts seem to be done because the extruder steps were so small that rounding error appeared.

rtyr, 2021-07-11:
Please provide your 3MF project file.

CZDanol, 2021-07-11:
[keyOrganizer_kk.zip](https://github.com/prusa3d/PrusaSlicer/files/6797558/keyOrganizer_kk.zip)

rtyr, 2021-07-12:
We should likely add some sanity check.

closed 2026-09-29 by prusaslicer-bot[bot]