prusa3d/PrusaSlicer #6701: Problem with PrusaSlicer

TheWebExpert1, 2021-07-11:
### Version
2.3.2

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa III3MK3, modified to 3S; firmware 3.10.0

### Behavior
PrusaSlicer has encountered an error.  It was likely caused by running out of memory.  If you are sure you have enough RAM on
your system, this may also be a bug and we would be glad if you reported it. bad allocation
I'm slicing a model I've sliced before.
[image]

rtyr, 2021-07-11:
Is your operating system 64bit? Are you running 64bit version of PrusaSlicer?

There was a bug in recent installer. It incorrectly installed 32bit version even on 64bit systems.

32bit OS can allocate only limited amount of memory.

TheWebExpert1, 2021-07-11:
Yes, I'm 64; slicer is 32

JKinch03, 2021-07-12:
Getting this same issue as well, just with the SL1. Not sure what's causing it, but it either gives me an error about memory usage, or the one shown above, or just crashes. It might be an issue with how the new version of the slicer processes models. I sliced something in 2.3.0, and 2.3.2, and it had no problems on my laptop in 2.3.0, but it wouldn't even slice it in 2.3.2.

rtyr, 2021-07-12:
If your OS is 64bit, please install 64bit PrusaSlicer. Fixed installer should be already available for download.

If it helps, please close this issue.

rtyr, 2021-08-09:
No response. Closing.

closed 2021-08-09 by rtyr