prusa3d/PrusaSlicer #6702: Error when slicing large model

Olvevitters, 2021-07-11:
### Version

Version 2.3.2+win32

### Operating system type + version
W10 64Bit home

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK2S
### Behavior
Was trying to slice the main body of the Swole Charmander at 287% 0.1mm detail, 30 % fill and support everywhere. The Software is saying i am running out of ram. Seems like it fails at the final stage, as i am able to see the model with supports however i just cant finish. Tried in Admin mode and without it

_Is this a new feature request?_ NO

[image]

rtyr, 2021-07-11:
You should use 64bit version of PrusaSlicer.

There was a bug in recent installer. It incorrectly installed 32bit version even on 64bit systems. Fixed installer should be already available for download.

32bit OS can allocate only limited amount of memory.

Olvevitters, 2021-07-11:
Okay i will try that

Olvevitters, 2021-07-11:
Worked with 64 bit

closed 2021-07-11 by Olvevitters