prusa3d/PrusaSlicer #6705: crash

vyachiks, 2021-07-12:
when i chose 100% infill (заполнение) in russian PrusaSlicer programm crashed...

[Window Title]
Critical error

[Main Instruction]
PrusaSlicer has encountered a localization error. Please report to PrusaSlicer team, what language was active and in which scenario this issue happened. Thank you.

[Content]
The application will now terminate.

boost::bad_format_string: format-string is ill-formed

[OK]
version 2.3.2 + win64

rtyr, 2021-07-13:
I was not able to reproduce it on 2.3.2/master.

bubnikv, 2021-07-13:
Please attach a 3mf for which setting the infill rate will cause the issue.
Thank you.

rtyr, 2021-07-13:
Could you please also provide information about your Windows OS (version, language)?

vyachiks, 2021-07-13:
Windows 10 64 pro, Russian language

Отправлено из myMail для iOS

vyachiks, 2021-07-13:
This only with Russian Ui… if I choose infill 99% program not crashed

Отправлено из myMail для iOS

vyachiks, 2021-07-13:
it happens anyway. even without loading a model for slicing. it is enough to select 100% filling in any menu and the program crashes. but if you select 99% then everything works fine

Отправлено из myMail для iOS

vyachiks, 2021-07-13:
Windows 10 pro 64edition Russian

Отправлено из myMail для iOS

YuSanka, 2021-07-13:
We can't reproduce your issue. 
[image]
Are you sure, that you use [official released PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.3.2/PrusaSlicer-2.3.2+win64-202107080703.zip) ?

vyachiks, 2021-07-13:
Yes, this is  <https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.3.2/PrusaSlicer-2.3.2+win64-202107080703.zip> official released PrusaSlicer !

 

 <https://user-images.githubusercontent.com/19184891/125408216-f643e900-e3ba-11eb-8abe-c02d154ccae9.png>

bubnikv, 2021-07-14:
@vyachiks

Please attach a 3mf for which setting the infill rate will cause the issue. We need as simple and reliable steps to reproduce your issue as possible.

Thank you.

Прикрепите 3mf, для которых установка скорости заполнения вызовет проблему. Нам нужны как можно более простые и надежные шаги для воспроизведения вашей проблемы.

Спасибо.

bubnikv, 2021-07-14:
I tried to switch the locale to Russian and I still cannot get the error message.
https://www.isunshare.com/windows-10/change-system-locale-in-windows-10.html

Maybe your installation is corrupted and you have the localization dictionaries mixed up. What happens if you download 2.3.2 from
https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.3.2/PrusaSlicer-2.3.2+win64-202107080703.zip

unzip it and run it from there? Then it should use the correct dictionaries.