prusa3d/PrusaSlicer #6708: Can't compile MASTER branch...

3Delights, 2021-07-13:
Just tried to compile the Master branch (Using Windows 10 and VS2019CE., as per the WIKI) and after running:

`cmake .. -G "Visual Studio 16 2019" -DCMAKE_PREFIX_PATH="c:\src\PrusaSlicer-deps\usr\local"`

I get the error message:
```
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
PSAPI_LIB
    linked by target "PrusaSlicer" in directory C:/src/PrusaSlicer/src

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

3Delights, 2021-07-15:
Working again.

closed 2021-07-15 by 3Delights

xiefeihua, 2022-11-18:
hello,  i have the same error .   but  "working again' don't work for me .  could you show me more detail for this problem,

yangminglong, 2022-11-30:
hello, i have the same error . but "working again' don't work for me . could you show me more detail for this problem,

yangminglong, 2022-11-30:
> D:\Dev\prusaSlicer\src\PrusaSlicer\build>cmake .. -G "Visual Studio 16 2019" -DCMAKE_PREFIX_PATH="D:\Dev\prusaSlicer\src\PrusaSlicer-deps\usr\local"
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_DESKTOP_INTEGRATION: ON
-- SLIC3R_ENC_CHECK: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_UBSAN: OFF
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: D:\Dev\prusaSlicer\src\PrusaSlicer-deps\usr\local (from cache or command line)
/Include//winrt/windows.graphics.printing3d.h was not found
STL fixing by the Netfabb service will not be compiled
Building without Win10 Netfabb STL fixing service support
-- Found Boost: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.66.0") found components: system filesystem thread log locale regex chrono atomic date_time iostreams
-- Boost::boost exists
-- TBB found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/TBB
-- Found ZLIB: optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/zlib.lib;debug;C:/Program Files/zlib/lib/zlibd.lib (found suitable version "1.2.11", minimum required is "1")
-- CURL found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/CURL
-- EXPAT found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/expat-2.4.3
-- Found ZLIB: optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/zlib.lib;debug;C:/Program Files/zlib/lib/zlibd.lib (found version "1.2.11")
-- GLEW found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/glew
-- try find NLopt in NLopt_DIR:''
-- Found NLopt in 'D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib'.
-- Using NLopt include directory 'D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/include'.
-- Using NLopt library 'D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/nlopt.lib'.
CMake Warning at cmake/modules/OpenVDBUtils.cmake:129 (message):
  vdb_print not found! OPENVDB_PRINT-NOTFOUND
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:317 (OPENVDB_ABI_VERSION_FROM_PRINT)
  CMakeLists.txt:555 (find_package)

>CMake Warning at cmake/modules/FindOpenVDB.cmake:324 (message):
  Unable to determine OpenVDB ABI version from OpenVDB installation.  The
  library major version "8" will be inferred.  If this is not correct, use
  add_definitions(-DOPENVDB_ABI_VERSION_NUMBER=N)
Call Stack (most recent call first):
  CMakeLists.txt:555 (find_package)

>-- TBB found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/TBB
-- Found Boost: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found version "1.78.0") found components: iostreams system
-- OpenVDB libraries: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libopenvdb.lib;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libopenvdbd.lib
-- IGL NOT found, using bundled version...
CMake Warning at D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/CGAL/CGALConfig.cmake:92 (message):
  CGAL_DATA_DIR cannot be deduced, set the variable CGAL_DATA_DIR to set the
  default value of CGAL::data_file_path()
Call Stack (most recent call first):
  src/libslic3r/CMakeLists.txt:329 (find_package)

>-- Visual Leak Detector (VLD) is not found.
-- Using header-only CGAL
-- Targetting Visual Studio 16 2019
-- Target build environment supports auto-linking
-- Using VC toolset 142.
-- Generator uses intermediate configuration directory: $(Configuration)
-- Found Boost: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.48")
-- Boost include dirs: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/include/boost-1_78
-- Boost libraries:
-- Adding precompiled header pchheader.hpp to target libslic3r.
-- WXWIN environment set to:
-- wx libs: debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_html.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_html.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_adv.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_adv.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_gl.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_gl.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_core.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_core.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxbase31ud.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxbase31u.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxregexud.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxregexu.lib;opengl32;glu32;winmm;comctl32;uuid;oleacc;uxtheme;rpcrt4;shlwapi;version;wsock32;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libpng16_static.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libpng16_staticd.lib;optimized;D:/ [... cut ...]
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
PSAPI_LIB
    linked by target "PrusaSlicer" in directory D:/Dev/prusaSlicer/src/PrusaSlicer/src

>-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

I follow the instructions： [Step by Step Visual Studio 2019 Instructions](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md)