prusa3d/PrusaSlicer #6709: 'Empty layers detected, the output would not be printable' caused by generated supports

stephenguise, 2021-07-13:
### Version
Occurs on both 2.3.1+win64 and 2.3.2+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 v2 (probably irrelevant)

### Behavior
* _Describe the problem_
An error is shown when slicing the file, though the error does not occur if supports are not generated. Changing some very small parameters in the file also causes the error not to occur.

* _Steps needed to reproduce the problem:_
1. Load **tabtest_v7_0_28.stl**
2. Relevant slicer settings:
Layer height 0.28 (initial layer height 0.20)
Supports: autogenerated
Contact Z distance: 0.15mm
Pattern: Rectilinear
Pattern spacing: 1mm
Interface layers: 2
Interface layer spacing: 0.2mm
XY separation: 500%
Don't support bridges: checked
3. Click 'slice now'

* _Expected Results_
Model slices correctly
* _Actual Results_
Error is shown: 
'Empty layers detected, the output would not be printable
Print z: 10.270000'
Scrolling the Preview pane to 10.27 shows some generated support structure will be printed at that Z-height.

NOTES: 

The very similar file **tabtest_v7_0_56.stl** is almost identical to the file which causes an error, but this file slices fine. The cutout section on which the support rests is lower by exactly 1 multiple of the layer height. The generated support is the same size and printed the exact same Z distance above the main structure, but no error is displayed.

The error does not occur if supports are not generated.

#### Project File (.3MF) where problem occurs
I have uploaded a zip file containing:
tabtest_v7_0_28.stl (original STL with which the error manifested)
tabtest_v7_0_28.3mf (PrusaSlicer export of the above STL after slicing)

tabtest_v7_0_56.stl (similar STL which does not cause any errors)
tabtest_v7_0_56.3mf (PrusaSlicer export of the above STL after slicing)

An export of the project from fusion 360, if necessary (you can simply adjust the parameter 'vcl' from [layer_height * 2] to [layer_height] or any other value and export the mesh to see the difference.
[tabtests.zip](https://github.com/prusa3d/PrusaSlicer/files/6812215/tabtests.zip)

bubnikv, 2021-09-22:
Would you please test with PrusaSlicer 2.4.0-alpha1? It seems the issue is gone, likely due to the support generator fixes.

NHodgesVFX, 2021-11-11:
@bubnikv 

I just hit this on PrusaSlicer 2.4.0 Beta 1

I was only able to hit the when using a .25mm nozzle and slicing for it. Although at one point when slicing for a .4mm nozzle I did receive an error about the minimum extrusion width being too small but I was not able to reproduce it consistently

Also, it would be good to add a clear all notifications button as this can generate 20+ notifications. Like the OP, the sliced file prints fine and it only shows up when Prusa Slicer generates supports. Slicing the model with no supports does not cause any errors.

[Suzanne.zip](https://github.com/prusa3d/PrusaSlicer/files/7517074/Suzanne.zip)

Steps to Reproduce:

1. Setup to slice for a .25mm nozzle on the Prusa Mini
2. Import the provided "Suzanne.stl" Into Prusa Slicer
3. If a popup complains the model is too small, click yes to re-scale it
4. On the right UI pane set Supports to Everywhere
5. Click Slice

Result: A bunch of error messages pop up complaining about empty layers
Expected Result: The model is sliced and no errors popup

Windows 10 64Bit
Prusa Mini + 0.25 Nozzle

rtyr, 2021-11-11:
I was able to reproduce it with "Suzanne.stl" (only with 0.05mm layer height).

3MF:
[MINI_Suzanne.zip](https://github.com/prusa3d/PrusaSlicer/files/7518878/MINI_Suzanne.zip)

lukasmatena, 2021-11-22:
@NHodgesVFX @stephenguise
I reproduced both issues in 2.4.0-beta1, but not in current master. I recently added a fix for the empty layer warning (aae8c7a844de7656b6d8f7a2feff0fb759288554), apparently it fixed your issues as well.

Closing. In case you see this again in the upcoming 2.4.0-beta2 (or later), definitely let us know. Thanks.

closed 2021-11-22 by lukasmatena