prusa3d/PrusaSlicer #6711: Filament overrides on saved 3MF MMU profiles are broken

Bendermk6, 2021-07-14:
PrusaSlicer
Version:   2.3.2+win64
Build:     PrusaSlicer-2.3.2+win64-202107080703

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition

Printer: PrusaMK3S+ with MMU2s

Behavior
Saving a MMU project with unchanged filament settings -> all filament overrides are OFF.
[image]

Opening the same 3MF file -> "Filament retract on layer change" and "Filament wipe while retracting" overrides are ON.
[image]
[image]

3MF Files
[mmu2_test2.zip](https://github.com/prusa3d/PrusaSlicer/files/6814242/mmu2_test2.zip)

rtyr, 2021-07-14:
I was able to reproduce it with 2.3.2. Not with 2.3.1 or current master.

bubnikv, 2021-07-14:
Thanks for the report. It is an unfortunate bug introduced in 2.3.2 with e7a6323470940c23a66efe773ed8ba01e15b28cb
    Configuration compatibility - implemented substitution and reporting for vectors of bools (including the nullable bools).

I have fixed that on the stable branch with 16e41ab819a3938c3f1b814275afc654f63add1a
We need to check the impact of the bug, whether it is necessary to roll out a patch release.

bubnikv, 2021-07-14:
I did an investigation of the impact of the issue. It only affects multi-material prints (with multiple extruders). Everything works well until one loads a configuration file or 3MF with multiple extruders defined. Then the two "optional" boolean values at the "Filament Overrides" get messed up: The 1st extruder is fine, the 2nd extruder will have the two "optional" values set to "off", the 3rd extruder will have them set to the value of the 2nd extruder and so on.

bubnikv, 2021-07-16:
Fixed with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.3

closed 2021-07-16 by bubnikv