prusa3d/PrusaSlicer #6712: Default Filament and Print Profiles (Per Printer)

Bupher, 2021-07-14:
### Version
2.3.2 win64

### Operating system type + version
Win 10 Build 19043

### 3D printer brand / version + firmware version (if known)
Sidewinder X1/Ender 3 (Not specific to a printer)

### Behavior
Cannot set default print and filament settings for printers. When switching between multiple printers, it does not switch to the default print and filament settings I would like to use. I tried to set this in the default filament and print settings options in the printer config, but it has no effect. I tried the profile name and the profilename.ini.

_Is this a new feature request?_

Maybe. It seems it was a feature at some point but was either removed or broken.

#### Project File (.3MF) where problem occurs
Not specific to a project 

Relevant section of my Artillery Config:
[image]

Relevant section of my Ender 3 Config:
[image]

Version Info:
[image]

Config Bundle:
[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/6816392/PrusaSlicer_config_bundle.zip)

rtyr, 2021-07-15:
This will work: (removed .ini and quotation marks)

```
default_filament_profile = Artillery PLA 210-60 0.8
default_print_profile = Artillery X-1 0.40mm SUPER SPEED
```

Please note, that the behavior related to switching between profiles has changed since PrusaSlicer 2.3.0-alpha1, so it may work differently than you expect. 

From 2.3.0-alpha1 release notes:

> When switching the Printer profile, the active Print and Filament profiles may no more be compatible with the new Printer profile. The old PrusaSlicer selected a Printer and Filament profiles marked as "default" in the activated Printer profile. The new PrusaSlicer selects the new Print profile with the following priority:
> 
> 1. Compatible Print profile with the same logical name aka alias.
> 2. Prefered Print profile with the same layer height.
> 3. Compatible Print profile with the same layer height.
> 4. Prefered Print profile.
> 5. Any compatible profile.
> 
> Thus when switching the printer from "Prusa i3 MK3" with the "0.2mm SPEED" Print profile active to "Prusa i3 MK2", "0.20mm 10mms Linear Advance" Print profile is newly activated, while the previous PrusaSlicer would select the default, which is "0.15mm OPTIMAL". The same logic applies to the selection of Filament profile, while the symbolic "filament_type" parameter is matched instead of the layer height.

Bupher, 2021-07-15:
I changed it back to no quotes and no .ini, but now I am confused. Your pseudocode doesn't ever mention the default profile stuff, it just has its own metrics to determine which profiles to switch to. Does anything ever use the default profile settings? It doesn't even appear to be used on boot, as it always just grabs the last filament and slicing settings I used.

I have my printers setup with different nozzle diameters, and I run a different style of filament in each of them. So when I switch between them, I don't want it to grab the same filament, even though it is compatible, and I don't want it to grab the settings with the same layer height, even though I have a profile set up for it. Is there any way to accomplish this with the currently available options?

Also, you mentioned only using compatible profiles, is there a way I can tell set what profile is compatible with what printer myself?

rtyr, 2021-07-15:
> I changed it back to no quotes and no .ini, but now I am confused. Your pseudocode doesn't ever mention the default profile stuff, it just has its own metrics to determine which profiles to switch [...]

It works as mentioned in release notes (prefered profile = default profile).

> Also, you mentioned only using compatible profiles, is there a way I can set what profile is compatible with what printer myself?

Yes, I recommend to do so. Go to Dependencies -> Compatible printers condition (available in print and filament settings).

For example, condition like this will make profile visible only for Artillery printer with 0.8mm nozzle (the word "ARTILLERY" has to be included in printer notes). You can see various conditions like this in bundled profiles.

[image]

Bupher, 2021-07-15:
Wonderful, thank you. I think I can make this work for now. I would still like a way for preferred to be the 1st option in the list, not the last, but this should make it better.

rtyr, 2021-07-20:
Current behavior makes more sense for unsaved changes dialog, that was the main reason why it was changed. See 2.3.0-alpha1 changelog. I am quite sure there is already feature request for the ability to set default filament. Closing.

closed 2021-07-20 by rtyr