prusa3d/PrusaSlicer #6713: Walls present in the STL should never disappear completely no matter how thin they are.

yoyo42, 2021-07-14:
### Version
Version 2.3.2+win64

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
i3 mk3s

### Behavior
In the manual it says:

Detect thin walls
By default, every wall consists of outside and inside perimeters (a minimum of two perimeters for thin walls). If there is enough space for it, an infill pattern is used to fill the void between these inside/outside perimeters. 

Enabling Detect thin walls lets PrusaSlicer generate just a single perimeter that works as both inside & outside shell. This will help to capture tiny details. **However, walls thinner than a single perimeter extrusion width will likely still be ignored.**

I would like the behaviour to be that if a wall is present in an STL then it is _always_ guaranteed to be present in the final model, even in the model wall is 0.0001mm for some reason and the nozzle is 0.8mm. There is never a case where I want thin walls to disappear completely, having them become one layer thick as a minimum is much less astonishing to the user.

rtyr, 2021-07-14:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3140. Closing.

closed 2021-07-14 by rtyr

yoyo42, 2021-07-14:
Fair enough, it is a duplicate. The other one was mine too, two years ago.

There are a lot of things tagged with 'gap fill', is that going to lead to a mass cleanup of them at some point? It seems like there are two issues (at least) there: one of losing features from the STL, another where the path planning leaves holes that should be solid. For #3140 it would be best to not lose them in the first place rather than patch up small gaps afterwards.

rtyr, 2021-07-14:
Improvements related to "gap fill" labeled issues are planned. It's very complex topic.