prusa3d/PrusaSlicer #6717: Feature Requests independent outer nth perimeters control

Flambro, 2021-07-16:
I do a lot of engineering prints where structural strength is key. I also believe that the biggest holdup in FDM printing is the drastic reduction in strength between the layers compared to longitudinal. CNC kitchen has done a whole series about how to get maximum strength out of your parts. He was able to achieve layer bonding that was almost as strong as in the longitudinal direction. Unfortunately quality suffers greatly when using most of those techniques such as print at hotter temps, reduced fan speed, and oversized line width.

If we could control the settings such as temp, width, and fan speed for the outer n perimeters independently from the inner perimeters and infill, we could optimize the inner parts for strength and the outer perimeters for looks/dimensional accuracy. I don't think that would be a very difficult thing to implement and it would have a lot of application in part optimization.

rtyr, 2021-07-16:
> If we could control the settings such as temp

Not very realistic, it takes significant amount of time to stabilize the temperature (+imagine how many times per layer would temp typically change - more or less successfully).

> width

It is already possible to set extrusion width for internal/external perimeters independently.

> fan speed

https://github.com/prusa3d/PrusaSlicer/issues/2776

jschuh, 2021-07-16:
FWIW, I experimented quite a bit with varying hotend temp as described, and as rtyr suggests I just don't think it's possible to stabilize the temperature quickly enough to make any difference within a layer.

Fan control is a different story. I wrote PR #2921 (from the issue rtyr linked) and find it very helpful for enclosure printing and higher temp materials.

Flambro, 2021-07-17:
For many of my applications I have many perimeters (50+). Maybe it is possible for it to run a heat-up/cool-down perimeter where it just prints slower than usual to allow the temp to stabilize without oozing everywhere? I am just thinking out loud here. But for many things I don't care how long it takes, even if a 3 day print is tripled in print time. I do realize though that development needs to follow what is most useful for the most people not just what one guy on the internet thinks would be nice haha.

And for the width it would be nice if it could be a distribution. For example the outermost perimeter could be 0.4 then the next 0.5 then 0.6 and so on until the desired width. This would allow each subsequent perimeter to better fill the preceding one. If we jumped from say 0.4 all the way to 0.6 there would be voids in tighter radii that the 0.6 doesn't fit but a 0.5 would. Voids are not conclusive to strength. Even cooler yet would be if it would dynamically adjust the width to fit in areas down to the external perimeter width and up to a set maximum. Where it would extrude the max unless smaller is needed to fill voids or the for the external perimeter.

As for the fan part thanks for the references. I hope that gets implemented.

Thanks for the replies as well.

closed 2023-09-12 by rtyr